ramapcsx2/gbs-control #4: 640x480 resolution?

andreacampanella, 2018-03-11:
I'm using this on a CRT Monitor, at 640x480 generates scan lines but I cant enable that.

ramapcsx2, 2018-03-14:
The 640x480 presets outputs a full 480 lines, similar to line doubling.
There is no mechanism in there that enables scanlines and in fact, I haven't been able to do scanlines in software yet.
Hardware scanline generators should work though.

Nodoan, 2018-03-15:
At lower resolutions PC monitor CRTs will have scan lines naturally. I too use a VGA monitor and end up having scanlines at 640x480, but not just from this device. I also get scanlines at 640x480 when using older gaming systems I have setup. For example when playing theme hospital or rollercoaster tycoon direct from a retro gaming build scanlines are visible.

It's just how the CRT works.

ramapcsx2, 2018-03-15:
Yeah, I remember my CRTs :)
The thing is, I don't want us to confuse which "type" of "scanlines" are or are not shown.
If you connect a PC to that CRT and have it output 640x480, and that creates the "scanlines" you desire, then gbs-control should produce them as well, in the the 640x480 preset.
If gbs-control doesn't do that, check if your source console is generating regular 240p content (ie: is not an N64 or PS2 or some other, typically interlaced source).

ramapcsx2, 2018-03-19:
Closing this for now.

closed 2018-03-19 by ramapcsx2