ramapcsx2/gbs-control #7: OSD Support

bkoropoff, 2018-03-12:
This would be nice to have.  I tried the sequence of register writes suggested by the programming guide, but I haven't been able to get anything to show up.  If you have any tips for getting it to work I can finish up the implementation I started.

ramapcsx2, 2018-03-14:
Sure. Here's a couple very useful dev command chains :)
`
osd tests: s0s90s00s0s91s01s0s92s75s0s93sc6s0s94s01s0s95s22s0s96s06s0s90sd2
`
I think the order of operations was important here. If you follow the example of the doc by the letter, it should work. This command may work as is though.

`offsets higher + y offset + gain + adc150mhz: s5s06s3ds5s07s3ds5s08s3ds3s35sffs5s09s0fs5s0as0fs5s0bs0fs5s03s01s3s3as10`
Use this with a black (or very dark) picture from your console to make visible all the noise that's being fed into the system somewhere.

ramapcsx2, 2018-03-19:
If you get the OSD going, what would be your plans for it?
I found that it's too limited for anything really nice. It can display some icons and then a bar for "more" and "less", so pretty much just cosmetical stuff.

bkoropoff, 2018-03-19:
I have the tact switches on the GBS board connected to my Arduino.  Right now I have them hard-coded to change screen position in all 4 directions, but with the OSD menu I can support changing screen size.  Can brightness/contrast/hue can be supported as well, or have you disabled the image processing stages of the video pipeline entirely?  Obviously anything more complicated than this (e.g. changing preset or resolution) can't be done well through the OSD, but it's better than typing commands into a serial console.

ramapcsx2, 2018-03-19:
Ah okay, I can see where you're coming from.
I really recommend you try out an ESP8266 board, and the associated web UI.
Bonus: Tons more RAM, boatloads more flash, 160Mhz core (and from what I gather, IO as well) :)

Basic color settings you mentioned are all available on the later VDS registers, in the "dynamic range expansion" segments. Y Gain is on 3_35, for example. You want that for "brightness". The "U" and "V" controls are responsible for color mixing, but can also be used as color boost.

ramapcsx2, 2018-04-12:
Yeah, I did. The sequence I posted earlier produces the OSD on the top left.
Let me put it in the middle of the screen :)
s0s90s00s0s91s01s0s92s75s0s93sc6s0s94s01s0s95s22s0s96s40s0s90sd8

bkoropoff, 2018-04-15:
Fixed by #15, although more enhancements are possible.

closed 2018-04-15 by bkoropoff