ramapcsx2/gbs-control #8: Sega 32x doesn't sync.

Nodoan, 2018-03-15:
I've at this point tried several SEGA 32x systems (model MK-84000) and all of them don't sync. I can get an image for a split second right at the start. However the original firmware does find 32x sync and scales. Possibly different with the MK-84000A but don't have a model to test with currently.

ramapcsx2, 2018-03-15:
I'm sorry, I don't have a bug hunt chart ready yet. As I also don't have a 32x, I need you to help out more to narrow this down.

So I don't have an extensive chart but here's a start:
- connect to a Serial Monitor for debugging
- disable syncwatcher right after a reset or right in the code
- post a few lines of the "i" command readout, so I can see what the scaler does

Nodoan, 2018-03-15:
Here's 20 lines of the "i" command. I think this is right.

`h:213 v:56 PLL:2 status:C mode:0 ADC:0 hpw:0 htotal:1023 vtotal:0
h:213 v:56 PLL:2 status:C mode:0 ADC:0 hpw:0 htotal:1023 vtotal:0
h:213 v:56 PLL:2 status:C mode:0 ADC:0 hpw:0 htotal:1023 vtotal:0
h:213 v:56 PLL:2 status:C mode:0 ADC:0 hpw:0 htotal:1023 vtotal:0
h:213 v:56 PLL:2 status:4 mode:0 ADC:0 hpw:0 htotal:1023 vtotal:0
h:213 v:56 PLL:2 status:4 mode:0 ADC:0 hpw:0 htotal:1023 vtotal:0
h:213 v:56 PLL:2 status:4 mode:0 ADC:0 hpw:0 htotal:1023 vtotal:0
h:56 v:56 PLL:2 status:4 mode:0 ADC:0 hpw:0 htotal:1023 vtotal:0
h:212 v:56 PLL:2 status:4 mode:0 ADC:0 hpw:0 htotal:1023 vtotal:0
h:212 v:56 PLL:2 status:4 mode:0 ADC:0 hpw:0 htotal:1023 vtotal:0
h:212 v:56 PLL:2 status:4 mode:0 ADC:0 hpw:0 htotal:1023 vtotal:0
h:212 v:56 PLL:2 status:4 mode:0 ADC:0 hpw:0 htotal:1023 vtotal:0
h:212 v:56 PLL:2 status:4 mode:0 ADC:0 hpw:0 htotal:1023 vtotal:0
h:212 v:56 PLL:2 status:4 mode:0 ADC:0 hpw:0 htotal:1023 vtotal:0
h:212 v:57 PLL:2 status:4 mode:0 ADC:0 hpw:0 htotal:1023 vtotal:0
h:213 v:57 PLL:2 status:4 mode:0 ADC:0 hpw:0 htotal:1023 vtotal:0
h:213 v:57 PLL:0 status:4 mode:0 ADC:0 hpw:0 htotal:1023 vtotal:0
h:213 v:57 PLL:2 status:4 mode:0 ADC:0 hpw:0 htotal:1023 vtotal:0
h:213 v:57 PLL:2 status:4 mode:0 ADC:0 hpw:0 htotal:1023 vtotal:0
h:213 v:57 PLL:2 status:4 mode:0 ADC:0 hpw:0 htotal:1023 vtotal:0`

ramapcsx2, 2018-03-15:
Yeah okay, the sync processor is "blind" to this input.

Does it go through a LM1881 (or similar) sync stripper? Or is it direct csync from the console? What kind of cable do you use?

Nodoan, 2018-03-16:
Sync as composite video through a gscartsw which has a sync stripper in it. Not a super good cable. Running on original firmware does work with the same setup and the genesis the 32x is attached to works on both.

Nodoan, 2018-03-16:
Tomorrow I'll use the sync scanner to the normal genesis then disable it and plug in the 32x and see if it works.

ramapcsx2, 2018-03-16:
Loosing a color on just one console? Sounds like your connector or cable messing up.

The other problem now sounds more or less like a software bug to me.
The control software has gotten quite complex and I already find bugs left and right.
Maybe if you continue trying things, I can get a better idea of what's going on.

"OSR" or oversampling ratio is normally at 4x. Hitting that button reduces it to 1x or 2x, but that's just a developer thing really. It should always be at 4x.

Nodoan, 2018-03-16:
I actually just checked it in chain with and without the GBS control to my capture card. My first bet is indeed poor wiring or something of the like. Works to the capture card without it in the chain and before it did work with a RGB CRT which I can't test with currently. Did use two different cables.

My guess is the sync stripper in the gscartsw just isn't playing nice with the quality of the signal. Can't turn it off for VGA output. So I'll probably make an adapter to test it without as well as get some better cables. If it still does it I'll submit a report but there's things to rule out beforehand.

ramapcsx2, 2018-03-17:
Please leave OSR alone (at 4x). If you need to shift the frame or canvas, there are controls for it. You can shift the horizontal sync, the framebuffer, etc.

That green tint doesn't appear to be a short in your cables. I don't know what's going on there..

There is a bit of crosstalk between the Composite Video line and your RGB. You can see that in the diagonal lines in your reds. Your cable might not be shielded well enough, or the switch may have issues.

Nodoan, 2018-03-17:
Alright a few things, I already said why I used OSR instead of only the image moving tools. This is because part of the image was cut off, the actual image, not the overscan area, albeit just a pixel or two. When moved back using the horizontal sync caused a huge red bar to appear seemingly whenever it felt like it. I'm making steps to resolve it but have only really managed "image moves but pixels remain cutoff without the red" which is on any other overscan setting on at this point three segas. With OSR set to 1x the entire screen resolves properly without the red bar even in the overscan area. I said this because it was some really weird behavior and was worth a mention. 

[image] It's usually in the overscan area, but for the sega genesis actually cuts off one or two lines of pixels in the actual game.

I think it's directly related to the sega genesis overscan area, as it only appears when it's using specific "background" colors which change the overscan areas color. As sometimes the red line isn't there at all and it can be some other color on occasion as well. I'm more worried about how it overlaps part of the game sometimes and shifts the color of the pixels.

As far as the green tint goes, Like I said I'll doublecheck everything before actually submitting it, it was meant to be in passing until you thought I was losing a color and then I had to 'splain.  I was actually seeing some other image considerations that made me think similar. 

But as far as the diagonal lines go that's an internal SNES thing and not the switch or the wire. It has to do with what is probably the wierdest S-video setup I've seen. It's fixed with either a full RGB bypass including trace cutting, or another known fix with the internal RGB amp on the "snes mini" is bridging C28 which is Luma to ground. (kills composite video, for reasons) It's interesting that is affects even the systems composite sync line and will show up on all outputs otherwise, including RGB with Csync and very common to see it on streams and such on older models that don't require a mod for RGB out. 

Fixing that while keeping the internal RGB amp is one of my current projects actually, and the reason your seeing it is that there's a cap there and not a bridge currently that I ended up using to keep composite video and try to mitigate the issue. Which did work FWIW. Straight to a CRT they are pretty much invisible.

Anyway as before, I'll submit a report once I've ruled everything out with the SNES thing, internal and external. And see if I can figure out why x does y in the meantime.

ramapcsx2, 2018-03-17:
>>HB ST (memory): 630
HB SP (memory): 800

Is that still valid?
If so, the red line is garbage from an improperly set memory cut off region. You can probably fix this by moving the cut off a little. Try these commands, one at a time, then revert back:
_whbst 620
whbst 640_
Did this work? If not:   
_whbsp 810
whbsp 790_

One of these should fix the line. Take that and find the minimum change from the default that still fixes the line. Good luck :)

Regarding the SNES, I can't remember any S-Video related color burst bleed, but on some later 3 Chip models (especially S-APU mainboards), the pattern is indeed already coupling into the RGB inside the console. The only sure fix for that that I know of is lifting pin 3 on PPU2, which is the color burst clock (3.58Mhz).

bkoropoff, 2018-04-15:
There have been a lot of recent changes to SoG handling and image boundaries.  This probably needs to be re-tested.

ramapcsx2, 2018-04-15:
Yeah, this is definitely outdated now. 
It's possible that there's still issues, in which case @Nodoan should open a new ticket.

closed 2018-04-15 by ramapcsx2