ramapcsx2/gbs-control #9: White balance regression

bkoropoff, 2018-04-02:
Commit caad7d16154442ae00f6a30bc2eb4d84c80d31c6 causes white balance problems on my Genesis.  Changes in the dominant color on screen seem to cause the entire screen to become tinted.  This is particularly easy to notice in the title screen of Sonic 2.  When the background fades in from black, the white portions of the foreground dramatically change color.  I soldered capacitors into my video cable before discovering it was a software problem.

bkoropoff, 2018-04-03:
Yeah, I added the 75 ohm series resistor as well.  I didn't understand what I was doing when I first built the cable.  Not sure how I got away with not having them.

ramapcsx2, 2018-04-03:
Don't worry, it'll be fine. I've accidentally shorted some RGB pins on occasion, chip still worked :)

I have an NTSC Sega Genesis hooked up now. The RGB clamp position is just great in the 240p test suite, but in Sonic 2, the active video does indeed reach as far back as to reach it!
I'll push the obvious fix in a minute.

ramapcsx2, 2018-04-03:
Fixed in https://github.com/ramapcsx2/gbs-control/commit/0c68378694954c3e96e811305d86b832fc5d46a5

closed 2018-04-03 by ramapcsx2