ramapcsx2/gbs-control #10: Active sync locking moves effective vblank start and stop

bkoropoff, 2018-04-08:
I noticed this when experimenting with changing display vblank start/stop in order to crop the rather noisy overscan areas produced by the Genesis/Mega Drive VDP.  Since the "different frame sizes" feature of the TV5725 only changes the effective htotal without changing the display or memory vbst/vbsp positions, the active region of the display effectively shifts down by two scanlines (although the contents do not).  Depending on what the console puts in the vblank region, the effect is visible.

ramapcsx2, 2018-04-08:
Yeah, this is a huge pain actually. It's not only an issue vertically, but horizontally as well.
I tried to base the various input formatter (IF) offsets on something in the PAL / NTSC standards but there are many more variables at play and I don't have the extensive documentation required.
It would be great if I knew how to definitely avoid garbage pixels outside my defined active video region.

Up till then, I've got to rely on testing a few consoles every so often.

Right now, Genesis seems okay on the vertical.

bkoropoff, 2018-04-08:
Yeah, moving the display vblank start/stop positions does work as expected, but it makes the aforementioned shifting effect visible (e.g. if you perfectly crop the screen in Sonic to eliminate the blue border, two pixels of the border become visible again when active sync increases htotal).  I'm going to experiment with going back to changing the output timings directly instead of using "different frame sizes", since that will let me change vbst/vbsp to compensate for the shift.  I'm hoping it will still be compatible with your display since the maximum adjustment will be only 1 scanline thanks to the htotal calibration.

ramapcsx2, 2018-04-08:
My display appears to work with vline offsets in multiples of 2. A one line offset won't work.

For this issue, I would work with the auto IF offset, which changes S1_1E.
You might also have to change S1_1C.

ramapcsx2, 2018-04-12:
I can see the effect as well now. The "trick" is to simply shift the picture downwards with '/'.
The new branch solves this issue nicely :)

bkoropoff, 2018-04-13:
Fixed by #11

closed 2018-04-13 by bkoropoff