ramapcsx2/gbs-control #12: Syncing regression on NTSC SNES

bkoropoff, 2018-04-09:
Recent master versions have a lot of difficulty syncing with my NTSC SNES.  There's a lot of stuttering as frames are missed.  Eventually it stops updating the screen entirely.

This was introduced by b962fd9edd8bd

ramapcsx2, 2018-04-09:
I tested SNES specifically (as well as Genesis) to make sure Macrovision coasting wasn't required anymore.
So it's probably an issue with your sync signal and/or any processing done on it.
Can you describe your setup please? Especially important is any resistors in the sync line and if you use an LM1881, it's power supply (3.3 or 5 volts).

ramapcsx2, 2018-04-09:
If you use direct console csync or Composite / Luma, the SOG slicer level is now important again (not so much / at all with an LM1881).
It can be changed by 'wsog x', where 'x' can be 0 (lowest) to 31 (highest). The default is 10.

bkoropoff, 2018-04-10:
My cable is wired to SNES csync through a 330 Ohm resistor.

What resistance do you measure on your GBS between sync and ground?

ramapcsx2, 2018-04-10:
I measure 1k on a "2014" GBS and 500Ohm on a "2017" model.
Both have a single 1K resistor to ground installed, so no idea what's going on there.
Your SNES csync is probably too strong with those components.
I always recommend adding an additional 100Ohm to ground on the sync input.

bkoropoff, 2018-04-10:
Yeah, it seems that the sync termination on the GBS is questionable, though it was designed for arcade hardware.  The 330 Ohm serial resistor, which is recommended for SCART cables, is calibrated to bring TTL-level sync to ~1V peak-to-peak when the input is terminated to ground through 75 Ohms.  I noticed that my SNES actually outputs ~3V peak-to-peak csync, though.  I also modified my Genesis to output csync from the CXA1145 instead being driven directly from the VDP, so I have no serial resistor in the cable.  I'll have to play around with different input/cable resistors tomorrow evening to find something that works for both consoles.

ramapcsx2, 2018-04-10:
If you have an oscope: You want to aim for 0.3 to 1.0Vpp with the cables all connected.

Without oscope, you should install the recommended resistor: 100 Ohms to ground on the GBS sync input.
For a regular Genesis, csync is quite tricky to get right. Those require 470 Ohm in series, then a 10uF capacitor, then the ~ 75 Ohm to ground on the GBS (due to that extra 100 Ohm resistor).
But your Genesis is modified, so just the 100 Ohms will do.

There is another set of important parameters on S5_35 and S5_37. They handle sync pulse width margins.
Due to several sync input possibilities, this hpw can have a huge range.
I just found out that Genesis consoles can reach up to "200" on their weak and delayed csync lines.
A regular console would be more like "120".
I'll experiment with automatic settings for that.
It will have to be continuous, I think, because the console can switch the hpw. Genesis in 256px mode is such an example: Completely different pulse width than 320px mode..

In any case, if you install the 100 Ohm resistor and follow the cable diagrams for each console (or buy good premade cables), then the electrical side is fine.

bkoropoff, 2018-04-11:
OK, my SNES and Genesis both sync with 75 Ohm csync termination on the GBS.

closed 2018-04-11 by bkoropoff

ramapcsx2, 2018-04-11:
Great, that's a relief. I think I have a good grasp on the SP now.
The correct termination resistance should also improve the signal integrity a little, as it helps avoiding reflections / ghost signals.