ramapcsx2/gbs-control #19: tv5725.h definition

goutsune, 2018-04-14:
Not sure how to use GIT yet, so I'm filling this file with register descriptions according to register map file, I'll upload it section by section starting from S3

ramapcsx2, 2018-04-14:
Oh, why not use the Wiki then?
I made it open to edits by the public :)

goutsune, 2018-04-15:
https://github.com/ramapcsx2/gbs-control/wiki/tv5725.h aye, done for now.

ramapcsx2, 2018-04-15:
Thanks, I bet this is great fun! ;p

goutsune, 2018-04-15:
I'm not sure how this even works, but I'd like to write some kind of tool for converting linux modelines into timing values. 

By the way, as is see it now active frame lock adds/removes lines from scan area thus changing actual vertical frequency (which makes image shrink and stretch on CRT btw), but is it perhaps possible to fiddle with pixel clock somehow? For the sake of completeness I've also tried using htotal instead, but results were even worse.

ramapcsx2, 2018-04-15:
A modeline convert tool would be hard to design. The chip has special limitations, exceptions and features that aren't considered in a modeline.
For example, the output (VDS) clock is limited to a small selection, except if you use the recovered H-PLL clock as a source. Then it becomes a lot more complex.

So assume that you can't work with the clock parameter.
You could still figure out some 5275 parameters from a modeline, check the sketch for set_htotal and set_vtotal as an example of what's possible.

Once everything is set, you will get no picture or some form of mess.
This is because the IF needs to be corrected, the PLL brought in the right ballpark, the memory regions defined and the VDS memory blanks be appropriate.

So yeah, it'll be hard ;p

Active frame locking had just had a revision. Is it still causing issues on a CRT (and/or LCD)?
We know that it's kind of incompatible but it appears that only the strictest displays care enough.
For example, the image is stable and garbage free on my PC LCD.
If you can think of other methods to adapt the in and outgoing timings, we can certainly try.
(Fiddling with the ADC pixel clock would just distort the image and not change the timings. Fiddling with the output clock would be much worse than the current method.)

ramapcsx2, 2018-04-15:
Interresting.. using an AD9851 DDS?

The register map is coming along nicely :)

ramapcsx2, 2018-04-15:
There's a 5725 clone chip, called PVX522. 
This also has a register map and some options are named differently there, occasionally with better descriptions.
Dooklink's shared folder: https://drive.google.com/drive/folders/0B9y2RH4Lb3MZLUYtWGFOMUFVdlE
Worth a look if the original description just doesn't make sense!

closed 2018-04-15 by ramapcsx2

ramapcsx2, 2018-04-20:
@goutsune Do you plan on doing the other registers as well?

In any case, thanks for helping :)

PR #23 (2018-04-15): Use register interface for screen adjustment
No real functional change, just simplifies the code.

Closes #19 as well