ramapcsx2/gbs-control #26: NTSC Genesis alignment regression

bkoropoff, 2018-04-20:
Just recently noticed this after coming back to my Genesis after a while.  The screen is now shifted down substantially by default such that the blue border in Sonic is showing, and shifting it back upward with the OSD actually causes vblank-stop to wrap past 0, which then causes active sync locking to jiggle the screen.  I tracked it down to e3002827, specifically this change:

```diff
-    writeOneByte(0x37, 0x62);
+    // 0x37 has to be below the current average (real) hpw
+    // this also changes with different pll dividers (presets)
+    writeOneByte(0x37, 0x58);
```

Going back to 0x62 fixes it for me.  I'll experiment with other values.

bkoropoff, 2018-04-20:
Screen is still shifted down at 0x59.  Screen shakes up and down like crazy at 0x60 (why?!).  0x61 and up is stable and aligned correctly.  I feel like you have to be John Nash to see the patterns in how the register settings affect the 5725's behavior.

ramapcsx2, 2018-04-20:
Try changing S1_d0. Try these values: 0, 1, 0x10, 0x22.

Sync Processor stuff is closely coupled with IF stuff. I don't have it all figured out yet though, but with the new register templates, I'm getting there :)

ramapcsx2, 2018-04-20:
You don't want to try forcing the S5_37 value anymore. I have logic in place that will put it where it should be for what the SP sees.
The shifted image is something that happens later, most likely in IF.

Could you post a few lines of the info debug prints? Need to see how many vlines SP counts.

ramapcsx2, 2018-04-23:
Found a couple more issues that are related. Depending on your specific setup, the recovered vsync stop position may be wrong. This is the parameter that I can't yet figure out from the signal.
It sits on S5_40 and you may want to try moving it. Best if you place its start position 2 steps below your target as well (S5_3f), but it doesn't seem to matter much.

bkoropoff, 2018-04-27:
I'll have to update again and play with this stuff a bit this weekend.

ramapcsx2, 2018-04-27:
Would be great if you could check the register templates. It can write to a single bit fine, but reading one bit always returns the initial value somehow.
Well, that or I totally misunderstood how this is supposed to be used xD
https://github.com/ramapcsx2/gbs-control/commit/97ffabace2d155542dd4f846321d1245c1a5151a#diff-c6dbb996c79492f242ffc032772c3010R1970

If your Genesis still has the offset issue, and toying with S5_37 can fix it, please post an info line for both states (working / fixed).

bkoropoff, 2018-04-27:
> 97ffaba#diff-c6dbb996c79492f242ffc032772c3010R1970

I'll take a look at the template thing too.  Could be a corner case for 1
bit wide registers I need to handle.

bkoropoff, 2018-04-30:
OK, I updated to latest master and... Sonic isn't looking so great.

[image]

bkoropoff, 2018-04-30:
I think the problem with the template was specializing 1-bit registers to use `bool`.  The problem is that the type isn't guaranteed to behave correctly with values other than 0 or 1, so the compiler is probably optimizing it into garbage.  It would be possible to keep it by reworking the templates so they have separate "work" and "final types" that the result is cast to at the very end.

ramapcsx2, 2018-04-30:
Sonic: I thought this would happen on some GBS variants sooner or later :/
You need a different memory clock on your board. Try s0s40s7c for ~128Mhz.
Your options are 0x0c to 0x7c. Only a few of them will look good, with little to no wave patterns in the picture.

Templates:
I think I understood some words you said there ;p

bkoropoff, 2018-05-13:
I tried all values of PLL_MS other than 2, and they all have artifacts.  0 seems to be the best, but there is still some noise.  All modes have incorrect borders.

ramapcsx2, 2018-05-14:
S4_37 (+38) and S4_39 (+3A) define the length of memory blocks to fetch and the respective offset.
The more memory needed, the more likely it is that the RAM chip can't keep up (or some other, not yet reversed effect).
You could reduce both to 0xB4 (total, so the high bits are zero).
That will cut off the right side of the picture, but if it is a memory issue, the image will be clear.

ramapcsx2, 2018-05-16:
Looking back at the posted pic, this appears to be more of an VDS horizontal blanking start / stop position issue. I simply don't get why defining a huge chunk of htotal for "memory blanking" is necessary. All it does is have the potential to mess up either the start or stop position slightly, and cause those artefacts.

Try changing the start position ever so slightly:
- get current positions with "," (note HB ST (memory))
- write that value + 10 or -10, vary that a bit, using "whbst x", where x is your value

ramapcsx2, 2018-05-20:
I assume the latest code mostly fixes the issues mentioned here.

closed 2018-05-20 by ramapcsx2