ramapcsx2/gbs-control #28: Sync Locking via DEBUG_PIN seems malfunctioning.

xuserv, 2018-05-19:
Hello, there.

This is my first time writing a GitHub issue and I'm non-native English speaker also not a good programmer, please excuse me if I did something ignorance stuff and let me know so I can fix it.

A few days ago, a friend of mine introduce this stuff to me and it seems interesting so I built the hardware and try it out.

When I hook DEBUG_PIN of HCKJ 8200 (A clone of GBS-8200) to D11 of Arduino, the screen isn't looking so great.

Here is setup picture of it : 
[image]
[image]

and Screenshots and Logs (with DEBUG_PIN and without DEBUG_PIN) : 
------ [With DEBUG_PIN] ------
[image]
```
starting
resetDigital
using RGBS inputs

MCU: 16000000
..
try HD
try SD
New Input
NTSC timing 
SD mode
resetDigital
Base htotal: 1794
Candidate htotal: 1819
Candidate htotal: 1819
Candidate htotal: 1819
Candidate htotal: 1819
Best htotal: 1819
HB_SP preset: 1640 adjusted: 44
post preset done
----
htotal: 1819
VDS_HSCALE: 638
HS ST: 1398
HS SP: 1518
HB ST (display): 1304
HB SP (display): 0
HB ST (memory): 1328
HB SP (memory): 44
----
vtotal: 1000
VDS_VSCALE: 512
VS ST: 987
VS SP: 990
VB ST (display): 950
VB SP (display): 55
VB ST (memory): 19
VB SP (memory): 20
```
------ [Without DEBUG_PIN] ------
[image]
```
starting
resetDigital
using RGBS inputs

MCU: 16000000
..
try HD
try SD
New Input
NTSC timing 
SD mode
resetDigital
Base htotal: 1794
post preset done
Base htotal: 1794
Base htotal: 1794
Base htotal: 1794
sync lock failed, check debug + vsync wires!
----
htotal: 1794
VDS_HSCALE: 638
HS ST: 1398
HS SP: 1518
HB ST (display): 1304
HB SP (display): 0
HB ST (memory): 1328
HB SP (memory): 1640
----
vtotal: 1000
VDS_VSCALE: 512
VS ST: 987
VS SP: 990
VB ST (display): 950
VB SP (display): 55
VB ST (memory): 19
VB SP (memory): 20
```

ramapcsx2, 2018-05-19:
That's actually a great bug report!
I can immediately tell what's going wrong there, due to the image and the logs.

The problem is here:
`HB SP (memory): 44`
This is too far out, and I don't even know how that happened.

In any case, please keep the debug wire connected. Fetch the latest commit and try again.
The latest code uses more effort to produce allowed video timings and the base preset is closer to your source's "1819" htotal.

Is that an arcade board by the way? 1819 seems pretty high, meaning the source's vsync interval must be pretty low, maybe below 59Hz?

ramapcsx2, 2018-05-20:
Okay, great! The video timing correction routine now works correctly.

The glitch you get in your video is more problematic. It has to do with the sync processor and there are several things that can go wrong. Unfortunately, I don't have an answer yet that works with all sources.

In your Serial Monitor terminal, send this command and see if it helps:
`s5s38s03`
or 
`s5s3es00`
or
`s5s56s05`

Maybe these commands will fix the issue by themselves, maybe you need to combine them. Maybe you need to try different values of x for "s5s38s0x".
Most likely one of these variables will fix the issue.

I'll continue reverse engineering the problem, so that we'll have a universally working solution some day.

xuserv, 2018-05-20:
Now everything seems fine with ```s5s3es00``` except up-down borders!
I will try with other arcade boards and get back with results.

Here is Video and Log : 

[YouTube Video](https://youtu.be/aP62C0tQb6o)

```
starting
resetDigital
using RGBS inputs

MCU: 16000000
active FTL: 0
..
try HD
try SD
New Input
NTSC timing 
SD mode
resetDigital
Base: 1802 Best: 1824 Diff: 22
post preset done
segment: 5
register: 62
was: 10
is now: 0
----
htotal: 1824
VDS_HSCALE: 638
HS ST: 62
HS SP: 182
HB ST (display): 1823
HB SP (display): 502
HB ST (memory): 1782
HB SP (memory): 342
----
vtotal: 1000
VDS_VSCALE: 512
VS ST: 987
VS SP: 990
VB ST (display): 950
VB SP (display): 55
VB ST (memory): 21
VB SP (memory): 22
```

Thanks for your hard work!

ramapcsx2, 2018-05-20:
That's great, so it was the "overflow protect" feature. I wish the chip developer had documented their features better :p 

In any case, I also worked on this and found out that I don't need an SNES specific workaround any more.
Could you test the latest update? Hopefully this fixes your issue as well, with no need to change S5_3E.

Tip:
You can change the image position via the Serial Monitor terminal as well.
The keys to send are "*", "/" (for up and down) and "+", "-" (for left and right).
Sources can position the image where they want, so each different source always needs a slight adjustment.

xuserv, 2018-05-20:
Awesome, updated to latest one and it seems great!

left, up, down borders are still a problem even adjusting image position but it could be my monitor problem so I will try with other monitors as well.

Display examples : 
NEO-GEO MVS (MV-1B) [KOF2003]
[image]
Atomiswave [Metal Slug 6]
[image]
System Board Y2 [KOF2002UM]
[image]
PGM2 [Dodonpachi Dai-Ou-Jou Tamashii]
[image]

ramapcsx2, 2018-05-20:
Those are great results. I didn't know whether arcade boards would work at all, so that's good news that they do :)

Many monitors have an auto adjust feature for the VGA port. Maybe this is enough to center all your games without even needing to correct it in the scaler.

I'll close the issue as fixed :)

ramapcsx2, 2018-05-20:
Actually, no. This is a good reference in case others have similar issues.
Re-opening this.

xuserv, 2018-05-21:
Okay, removed C11 capacitor and some visual noises in the white colored background are now gone! neat.

And now, ToaPlan 2 [Snow Bros. 2 - With New Elves] PCB sync is stable!
Original firmware didn't support this properly.

About Left, Up, Down borders, auto-adjust on monitor did help Left borders but Up, Down borders are still in there.
I think some monitors does not accept 1280x960 output properly, so will do with 720p preset and get back to you with results and logs.

Display example : 
ToaPlan 2 [Snow Bros. 2 - With New Elves] / Test Mode
![ToaPlan 2 [Snow Bros. 2 With New Elves]](https://i.imgur.com/7dQ4vBe.jpg)
Taito FX-1B [Ray Storm] / Test Mode
![Taito FX-1B [Ray Storm]](https://i.imgur.com/0JBO4Yv.jpg)

ramapcsx2, 2018-05-21:
The ToaPlan 2 board almost doesn't work. You can see the top few scanlines are a bit warped.
This is a sign that the H-PLL has issues locking right after VSync end.
Maybe this board wants to see a different coasting setup.
(Coasting is the amount of lines before and after VSync where the H-PLL ignores the source pulses and "coasts" on its internal clock. This is required because on most systems, there are erroneous pulses around that time.)
Coasting is setup in S5_38 (pre coast) and S5_39 (post coast).
If you want, you could try changing the values and see if it improves:

`s5s38s01 s5s39s03`
- sets pre coast to 1 and post coast to 3
- usable values are 1 to 10 for both

tracedebrake, 2018-09-29:
Hello @xuserv & @ramapcsx2!

Might I ask, @xuserv, if you have done every recommanded modifications to your GBS 8200, including soldeging a 100ohm resistor from CSYNC to ground & setting all 3 RGB potentiometer to the minimum (left)? 

I'm asking this because normally, most arcade [SuperGuns](https://en.wikipedia.org/wiki/SuperGun) will not **[properly](https://etim.net.au/av-driver/arcade/)** adjust your RGBS signals to be 75 ohms & 0.7v peak-to-peak levels compatible. Arcade video signals have the same timing as NTSC video but the electrical properties are different to a standard TV/video monitor. It's analog signals with **TTL level amplitude (3-5 Vpp)** while regular gaming consoles are made for home use so they output at **0.7v pk-to-pk for a 75ohms load** (your tv) . Arcade monitors also have a very high inputs impedance to match those TTL voltage levels (1k-10k ohms).

This is why the GBS8200 has a high input resistor (1k) on CSYNC; it is made for arcade use by connecting your JAMMA harness directly to the RGBS pins.

By doing the recommanded mods (100ohm resistor + adjusting your pots total left (or removing them)), what you are actually doing is modifing your GBS8200 to be console compatible. If you want to use a SuperGun with a fully modified GBS8200, you need to make sure 1st that your [SuperGun properly adjust the RGBS signals to TV levels using a buffer/driver](https://www.arcade-projects.com/forums/index.php?thread/4904-need-a-hand-with-supergun-video-attenuation-circuit/&postID=70658#post70658), such as the [HAS v3.1](https://www.arcade-projects.com/forums/index.php?thread/332-home-arcade-system-new-supergun-sets/&postID=2521#post2521)

ramapcsx2, 2018-09-29:
Well, that is worrying..
So what do these Supergun things do exactly?
They're quite large boards with many components. I always assumed they are meant to adapt an arcade board to home TV equipment.
Outputting 4 or 5 Vpp on RGB seems silly enough, but then putting those signals onto a SCART socket?
I don't get it..

ramapcsx2, 2018-10-12:
@tracedebrake 
May I copy paste your comments to the Wiki here?
I'd like to document this for other Arcade users, so they don't blow up the nice ADC on the GBS ;p

@xuserv 
Great that you got it working!
I don't support Arduino (Atmel) anymore. The project outgrew the limited program memory.
Just a heads up ;p

tracedebrake, 2018-10-13:
@ramapcsx2 Yes sure, you're welcome! Still haven't received my new GBS-8200 yet, i'm planning to work on the lm1881n/rgb inputs mods dilemma as soon as I get them.

ramapcsx2, 2019-08-02:
The features discussed in this issue work much better now. Time to close it.

closed 2019-08-02 by ramapcsx2

Slypty, 2021-11-20:
Which Arduino board is that in the first pic?  I'm new to Arduino.  Received this clone board from Ali (HCKJ-200 v5.1) and I can't control the WiFi module ESP8266 to make a solid connection the gbscontrol connection.  I think the problem is the age of this clone with this WiFi module, so I'd like to downgrade to get this one going.