ramapcsx2/gbs-control #36: LG 1530s support

silensilen, 2018-06-21:
Hello!
First of all I wanted to thank you for your work.
I use your code to connect to the GBS-8200 Soviet retro computer BK0010-01 via RGBS
Everything works well. But the monitor of the LG 1530s does not support the output frequency, it reports the output from the range.
The parameters of this monitor are: lines: 31-63 kHz; frames: 56-75 Hz
Is it possible to correct the code, what would this monitor have earned?
Thank you in advance for your answer.

ramapcsx2, 2018-06-23:
Hi,

I need to see a picture of the BK0010, if you get it working! :)

Several output timings are available using the web interface (ESP8266 MCU only).
If you can't access the web interface, open a Serial Monitor (Arduino terminal) and send '9' for the 640x480 preset.

Good luck!

retroday, 2018-06-23:
Hey rama, this is the same problem I'm experiencing, that soviet computer outputs a PAL signal and that monitor unfortunately cannot display 50hz.

ramapcsx2, 2018-06-23:
Ah yeah, it would use PAL / 50Hz to be able to work on regular TV sets of the day.

silensilen, 2018-06-24:
I decided to use a TV with SCART instead of the GBS and the LG monitor. Now everything works.

closed 2018-06-24 by silensilen