ramapcsx2/gbs-control #37: Arduino nano 168 or 328P

tvirusx, 2018-06-28:
Hi there and first of all I would like to say that this project is awesome it has been a blast to use this board with great image and no issues so far, but I have 2 8200 board and I don't yet have the superpowers to code like u guys did on this, I've overlooked the code and with a small or little notion of program, could not reduce it to fit on an Arduino nano 168 nor 328. This is not an issue but a question. I tried to reduce the *.h files and lines that call for the ESP8266, but some of the errors that I get on variables I can not decipher what it is or how to work around it. I don't want to give any trouble with this as I have a normal Arduino and it is working great, but I have a second 8200 board and 2 Arduino nano's laying around and thought that I could use it. I really wish I could contribute to projects like this, but have no idea on how to learn to code on stuff like this. Really Great work u guys... keep it up
Thompson.

ramapcsx2, 2018-06-28:
The code is pretty bad, pretty much what you'd expect from an ever evolving project with no clear goal. 
As long as I keep finding new methods to use the scaler chip, the code will keep growing.
Some day a total rewrite is going to have to happen though.

In any case, to your questions:
The Nano 328 has similar specs to a regular Arduino Uno, which still barely fits.
For the 168 it doesn't look good though.

If you need to trim a few bytes to make it fit, you can delete some of the larger functions that you don't need. Example: getVideoTimings() is only called via console command. Deleting it frees 3% of program space.
If you need much more space, get rid of one of the feedback_clock presets. Delete the #include and bypass any code that uses the preset.

But the best solution to this is getting an ESP8266 board. They are readily available, compatible with Arduino and they fit this project much better. Also you'll get the web interface :)

goutsune, 2018-07-11:
Just built and uploaded master version onto my pro mini.
@tvirusx, I've managed to reduce code to 32078 bytes by commenting all webserver related functions and a few serial prints. While you can upload only up to ~30K with arduino bootloader you can still use full 32K by uploading with SPI programmer (e.g. with Arduino ISP if you have some UNO  to spare it's just perfect.).

tvirusx, 2018-07-12:
Yes, thanks very much for that additional info @goutsune. They way @ramapcsx2 said about getVideoTimings() worked just fine too, but with his new update, the full master file will not even fit the Uno anymore lol, but that's great stuff no worries there. Sorry, it took me so long to answer back, been a bit busy here... Another thing is that I am trying to get to work are the 4 buttons that he declares between lines 62 a 66. Am changing the pins to see if it works...

Thanks once again

ramapcsx2, 2018-07-13:
Hey,
I'm sorry the code doesn't fit. I'll try and slim it down a little over the next days.

If you want the buttons, you first have to wire them up (4 extra cables) to the mentioned IO pins.
Then uncomment #define HAVE_BUTTONS here:
https://github.com/ramapcsx2/gbs-control/blob/master/gbs-control.ino#L62

ramapcsx2, 2018-07-16:
The code fits onto a regular Arduino Uno / ATmega328 again. (The 168 progmem is just too small for the project.)

closed 2018-07-16 by ramapcsx2