ramapcsx2/gbs-control #39: GBS-82xx PAL/NTSC Tidbit/Observation

CrashSerious, 2018-07-12:
This isn't so much an issue, as an attempt to share a finding. (Hopefully it helps you, ramapcsx2.)

I have a GBS-8220 that I used for a NTSC Amiga 2000, it works flawlessly with a sharp picture. When I purchased a PAL Amiga 600 I was hoping to use it for that as well. It didn't work, the picture was garbled,  initially I thought that was because something was wrong with the Amiga 600 but Composite worked perfectly.

I then read about people having issues, and found this repository. I also had a couple GBS-8200's on order. When I tried one of them with the Amiga 600 it worked! (Picture is a little dark, bust that may be an adjustment).

So, I tried hooking the new GBS-8200 up to the Amiga 2000 and it DOES NOT WORK with NTSC Inputs.

From this observation, I suspect the GBS-82xx *MAY* set itself to PAL or NTSC at first power on or when it receives the first signal and store that in Flash.

ramapcsx2, 2018-07-12:
Hey,
the boards don't "remember" anything with the bypass jumper on.
You mention a garbled picture on board A but not on board B?
This sounds like different memory timings. Try cycling through the available memory clocks and see if it gets any better.

closed 2018-08-20 by ramapcsx2

CrashSerious, 2018-10-01:
Thanks for the response, and apologies for the delay on my side. I simply noticed it as an observation and thought I would share it after a bit of research. This repo seems to be the most advanced in terms of reverse engineering this marvel and truly opening it up, but I must admit as of yet I haven't invested the effort/time into getting it working. Two reasons; First being work/other pet projects and second being I know it would grab my attention and this is all I would do. ;-)

Planning to get back to it though soon.