ramapcsx2/gbs-control #42: Impossible to compile

CrossmanBx, 2018-08-04:
Hello
I've got a problem to compile your work with arduino IDE 1.8.5 and 1.9beta.
I Tried it with linux, and windows too.
I'm a newbie with WEMOS D1, and i got this problem .

Arduino: 1.9.0-beta (Windows 10), Board: "LOLIN(WEMOS) D1 R2 & mini, 80 MHz, Flash, 4M (1M SPIFFS), v2 Lower Memory, Disabled, None, Only Sketch, 115200"

sketch\PersWiFiManager.cpp.o: In function `PersWiFiManager::attemptConnection(String const&, String const&)':

sketch/PersWiFiManager.cpp:35: undefined reference to `WPS_is_unavailable_in_this_configuration__Please_check_FAQ_or_board_generator_tool'

sketch\PersWiFiManager.cpp.o:(.text._ZNSt17_Function_handlerIFvvEZN15PersWiFiManager17setupWiFiHandlersEvEUlvE1_E9_M_invokeERKSt9_Any_data+0x3e): undefined reference to `WPS_is_unavailable_in_this_configuration__Please_check_FAQ_or_board_generator_tool'

collect2.exe: error: ld returned 1 exit status

exit status 1
Error compiling for board LOLIN(WEMOS) D1 R2 & mini.

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

Thank you for your help

ramapcsx2, 2018-08-05:
I don't recognize this error and the fault described doesn't fit the code.

The string appears to be a library error message related to WPS.
I found this issue on it, if you need a lead: https://github.com/tzapu/WiFiManager/issues/650
Mind though that I'm not using "WiFiManager" but "PersWiFiManager", but maybe the suggested fix works.

The problem might be a recent update to the ESP8266 Arduino core?

ramapcsx2, 2018-08-05:
Note that the WPS feature is pretty silly anyway. It's not necessary in gbs-control.
If it helps, any code related to it could be removed.

CrossmanBx, 2018-08-08:
Thanks Rama for your answer.
It's ok now by removing these strings in PersWifiManager.cpp.

_server->on("/wifi/wps", [&]() {
    _server->send(200, "text/html", "attempting WPS");
    WiFi.mode(WIFI_STA);
    WiFi.beginWPSConfig();
    delay(100);
    if (WiFi.status() != WL_CONNECTED) {
      attemptConnection("", "");
    }
  }); //_server->on /wifi/wps

Thank you for your great job !
My Pc Engine Duo, Megadrive and Saturn look fantastic now on my old LCD !

ramapcsx2, 2018-08-08:
Yeah, I think I'll remove the feature, avoid this issue for everyone.
Enjoy your consoles, and nice to hear the PC Engine works :)

ramapcsx2, 2018-08-11:
Fixed in the development branch

closed 2018-08-11 by ramapcsx2