ramapcsx2/gbs-control #44: Having 640x480 preset active and then activating sync passthrough results in unstable/no image

Ryoandr, 2018-09-27:
Depending of monitor, result might be flickering image or no image at all ("no sync" osd)
Going from 1280x960, 1280x720, 1280x1024 to sync passthrough works, on one monitor at least.

ramapcsx2, 2018-09-28:
Okay, this should be fixed in development now.

Ryoandr, 2018-09-28:
It is almost fixed, I sometimes have to push the "source passthrough" button twice to get the display to work (regardless of current preset).

ramapcsx2, 2018-09-29:
Okay, what happens when it doesn't work? Any console output?
Each time bypass mode gets activated, it loads the default preset first now.
So each time should be the same really, unless there's a bug somewhere.

ramapcsx2, 2018-10-15:
Should be fixed due to various program flow improvements.

closed 2018-10-15 by ramapcsx2