ramapcsx2/gbs-control #45: Sync passthrough gives wierd results on a particular monitor

Ryoandr, 2018-09-27:
On a JVC DT-V, on cold boot, passthrough with MVS works but exihibits a smearing glitch on the right border, but after a while it starts getting more and more wavy, until it cannot sync anymore.
The other monitor still manages to sync, however the smearing is still present.

[image]
[image]
[image]

The DT-V monitor will only work cold for some reason. Note that it syncs reliably to MVS direct (but displays upper skew).
MD has the same behavior.

Ryoandr, 2018-09-28:
Latest dev seems to have fixed the sync issue, has been running for a while with no issues. Also horizontal position is pretty damn perfect now. The right smearing on MVS can be fixed by pushing the "Right" button twice in the "Move Picture (memory blank, HS)" section, so all is looking good.

ramapcsx2, 2018-09-29:
Might be even better now :)