ramapcsx2/gbs-control #46: Brightness boost on 640x480 preset scanlines is too big.

Ryoandr, 2018-09-27:
The right boost can be obtained if using scanlines on another preset and then going to 640x480 and activating scanlines.

ramapcsx2, 2018-09-28:
Yeah, I know that it only really works as intended with an unmodified preset and the ADC levels already correct for the source.
I'll have to think how to deal with this better.
In addition, scanlines only look good in the x1024 preset (at least here).
Because it looks bad in all the other presets, I haven't looked at the results there.

ramapcsx2, 2018-10-02:
The white line on the left was an easy fix. It'll be included in one of my next updates.

ramapcsx2, 2018-10-26:
This is now handled along with the deinterlacer, which in turn changes some rules for the scanlines.
I'll keep improving these enhancement options.

closed 2018-10-26 by ramapcsx2