ramapcsx2/gbs-control #47: Regarding GBS8200 mods

tracedebrake, 2018-09-29:
Hello!

In the ""Build the hardware page''

**Sync**

_''Every GBS board requires an additional 100 Ohm resistor on Sync-in to ground if you want to use the RGBS input.
Together with the factory resistor of 1000 Ohm, this brings the total resistance down to TV levels of 75 Ohm.''_

Wouldn't it be better to use a 82 ohms resistor? A 100ohms in parallel with a 1k makes a value of 91 ohms, which is 21% higher than the expected value of 75 ohms. By using a 82 ohms resistor, which is a regular value easily purchasable, it would bring the value at around 76 ohms.

**SCART RGB to VGA adapter**

_''If you want to build an adapter, here is some good inspiration: https://shmups.system11.org/viewtopic.php?f=6&t=35423''_

That setup uses the same LM1881N chip as the Sync stripper section but, there is no R1!: 

- _''For best results however, I recommend using a sync stripper to extract the various forms of RGBS sync.
Please ensure your circuit has a 75 Ohm to ground (termination) resistor on the LM1881 video input, as well as a 470 Ohm series (attenuation) resistor on the sync output. 

User viletim shows a good LM1881 circuit here:
https://shmups.system11.org/viewtopic.php?f=6&t=55948&p=1153713#p1153713 ''_

ramapcsx2, 2018-09-29:
Hi,
I'm really undecided on what I want to recommend for sync.
You are right, no schematic shows an input termination resistor for the LM1881.
I might have included it because I sometimes power the LM1881 from only 3.3V.
In that case, the input amplitude can't be too big, or else it fails to work.
But yeah, there's no documentation that has this termination resistor, so I should drop it.
Thanks for bringing it up! :)

Viletim's diagram is interesting, because it omits the Rset resistor and associated capacitor.
This circuit worked when I tested it, only the VSync output wasn't working.
Since VSync goes unused anyway, the simplified circuit can save a little work / space :)

The ~100 Ohm resistor (when not using an LM1881) has a few advantages over a perfect match:
- It's a very common part, easier to find than 82 Ohm
- Signal reflection from mismatched impedances is no concern in this setup, but a too weak amplitude will be! Better be a little on the safe side.
- Some GBS boards have just ~ 500 Ohm termination, others have ~ 1 kOhm. With the 100 Ohm recommendation, both types are fine.

ramapcsx2, 2018-09-29:
I've updated the sync section.

tracedebrake, 2018-09-30:
I'll look in this more closely since i'm not sure that removing R1 is the way to go. I'll print the datasheet out and use it as toilet reads. I'm also not sure that having a section for SCART RGB to VGA and one for SYNC just below is the way to go as well since they both do the exact same thing: sync stripping out of the RGBS. Maybe proposing a mod to do it on the board would be better since it would work for both scenario.

ramapcsx2, 2018-09-30:
xD
Well, I re-read viletim's explanations and they seem to be sound.
He does recommend installing R1, so that the current for sync can flow, but I'm not sure this is for the best in this setup.
I've noticed *decreased* LM1881 performance regarding jitter (measured by the amount the H-PLL on the GBS scaler stays locked), and some weak sources (Mega Drive) can even loose sync when terminated with low impedance (75 Ohm).

I'll try contacting viletim. Maybe he has some ideas on that.

tracedebrake, 2018-10-03:
That's some valuable informations, thanks. My research will be based initially on the V5 version. Since my new boards come from China, I can't really say when, But during november is my goal.

tracedebrake, 2018-10-03:
Looks like I accidently closed the issue; sorry about that. That's some valuable information, thanks. My research will be based initialy on the V5 revision. I do not know when the boards will arrive from China so it's hard to predict a date, but my goal is to share my findings during november.

ramapcsx2, 2018-10-05:
Best of luck, I hope you can find something to help with a proper input circuit :)

Viletim has replied, basically reconfirming that the termination resistor (75 Ohm) should be there.
If we do it this proper way, then the situation with Mega Drive's transistor CSync needs to be dealt with.
And of course, there's also the possibility that real HSync from a PC arrives on the "S" pin, as well as other console's CSYNC.
At least all other consoles I've tried deliver good CSync, as long as a ~470Ohm series resistor in the cable attenuates the signal.

CrossmanBx, 2018-10-08:
Hello rama,
i had to remove the 100ohm resistor between sync and ground, cause i had many lost of sync. Without it, it's perfect (saturn and megadrive). No problem with pc engine duo though.
It's very weird.
I'm going to test with a composite sync cleaned by a lm1881, soon, just for testing.

ramapcsx2, 2018-10-09:
Okay, if that works for you.
Remember to test if the resistor becomes required (or beneficial) when you update the software.
A lot of parameters are software controlled, and one day I'll get them all right.
You'll want to have the resistor then, as it actually is required.

(Unless you're using CSync directly, of course. This all only applies to CV or Luma Sync.)

ramapcsx2, 2018-10-10:
I'll just leave this here :)
https://shmups.system11.org/viewtopic.php?p=1337578#p1337578

tracedebrake, 2018-10-24:
Good news! I have received the 3x GBS-8200 that I ordered. To my surprise the 3 units are labeled as ''HD 9800 v5.0, 2015.01.15''

They are exactly like the following picture: https://i.ebayimg.com/images/g/4WMAAOSwfDhZhFxi/s-l1600.jpg

So far at 1st look they look fairly identical to the GBS-8200 v4.0: http://obligement.free.fr/gfx/gbs8200_1.jpg

All input connectors as well as the 3 pots are already desoldered on one board; I plan to work on this as soon as possible, mostly around mid-november. Will post my conclusion in this thread.

ramapcsx2, 2018-10-24:
Quite nice version of the board. You've got Samsung RAM and the 470uH inductor.
Cheaper (but not necessarily worse) boards use different RAM and only a 330uH inductor.
Overall though, these boards are all nearly identical.
Only the very first revision (with 2 outputs) and the HDMI versions differ more significantly.
The first revision might be better suited for generating Component Video output, since it comes with a video driver that can generate negative voltage (for the -0.3V sync tip presumably).

jejuho, 2018-11-04:
@tracedebrake did you manage to get this version of board working with gbs-control?

I have the same and I have it wired to Amiga 500 (GND + RGB + VS + CS with 680 ohm resistor in series). The "onboard" software works but I am not happy with the quality. I have wired wemos d1 mini to the I2C bus (D1&D2) and also the debug pin (D6).

If I short the P8, power on the HD 9800 V5.0 board I get no video output. The wemos/gbs-control is communicating via I2C bus, but I think it is not correctly detecting the input source:
`found: 10 getVideoMode: 0 input: 0
 YUV
`
This is always the result no matter what I try. So, what could I try to get this working?

tracedebrake, 2018-11-04:
Hello @jejuho,

Did you try something else than your amiga? A good test would be any component source you have at home laying around.

Can you reach GBS-Control over Wi-Fi?

jejuho, 2018-11-04:
Hello @tracedebrake,

not yet. I have a PS3 which I think can put out RGB. Would that work? I tried already with amiga's component but it would not work either. I measured with scope the debug pin signal with onboard SW and it is 50Hz as it should. With the external control I am not seeing anything except noise.

tracedebrake, 2018-11-04:
Caution: the component input on the GBS-8200 is for YUV/YPbPr. Do not plug a true RGB sync on green signal in there.

And can you reach GBS-Control over Wi-Fi?

CrossmanBx, 2018-11-04:
Hello rama
i've a lot of "found: 10 getVideoMode: 0 input: 0 YUV" too now, with a japan NTSC saturn since about 20 days (devellopment branch). GBS-control is unreachable while it occurs.

jejuho, 2018-11-04:
@tracedebrake  yes I have the wi-fi access. I have tried playing around with the options there but the screen stays dead. 

The PS3 approach was not successful. I could get crappy looking picture only (grayscale with noise/distortions) with the onboard SW. Nothing with external. I tried all the available options from PS3, the other one was RGB and other one maybe YpbPr (or YUV or something other). I am afraid I do not have any other sources to try.

jejuho, 2018-11-04:
I suspect my I2C communication is not working, even though the application is happy to push forward. The register values dump command 'd' shows register values as 0 to 5 based on the segment.

ramapcsx2, 2018-11-04:
Sync errors / no output:
Please run information mode via web ui (print infos) or serial monitor (send 'i') with the source turned off at first.
Then power up your source and grab what is happening with the info mode output.
This'll tell me what's going on internally.

An Amiga doesn't do any big tricks with sync, iirc.
Make sure the sync level is right by having an 680 Ohm inline resistor and the recommended 100 Ohm to ground extra termination resistor. Also use the RGBS input for this.

The PS3 using a Component cable should work easily (or anything else that outputs Composite for a TV).

These easier to work with sources failing hints at old preset files being loaded or similar.
You can try erasing them by uploading the sketch with the option "Erase Flash: All Flash Contents" selected.

ramapcsx2, 2018-11-04:
> The register values dump command 'd' shows register values as 0 to 5 based on the segment.
Yes, that's a failed I2C state.
Did earlier versions work? I recently changed the init routine to go forward even in that condition, to allow for WiFi config changes and setup even when the GBS is missing.
If you want to try an earlier version, use a master branch sketch from maybe a month ago.

The jumper to disable the original MCU is installed? I2C soldered or plugged?

ramapcsx2, 2018-11-04:
> Hello rama
> i've a lot of "found: 4 getVideoMode: 0 input: 1 lost" too now, with a japan NTSC saturn since about 20 days (devellopment branch).

Does it work when you send 'wsog 8' via serial?

jejuho, 2018-11-04:
@ramapcsx2 sorry for the trouble. It turns out it was a combined I2C & sync signal issue. I originally had the I2C wired right way around but the sync signal incorrectly. After that I switched the I2C wires the wrong way around... now both are wired correctly and all is good.

ramapcsx2, 2018-11-04:
No worries. 
I'm pretty sure this happens often, as there is little feedback when users first build this device.
Error messages and such exist, but people would first have to connect to WiFi or serial monitor to see them.

Maybe I should add some kind of elaborate LED blinking action? 
Would that get people's attention? :p

Biker888, 2018-12-11:
Hello !

I have a 8200 on a machine (so not gaming stuff)... it seems that the EGA video from 1989 is a bit wider than the 8200 wants to have it (i see a picture but the 1/10th of it is always ourside the frame)
Tried all resulutions , H size , H positions.. no luck.

Biker888, 2018-12-11:
[image]
[image]

ramapcsx2, 2018-12-15:
It'll be hard for me to support this from here.
Is this source connected via RGBHV or RGBS?
If RGBS, try using bypass mode (selectable via web ui).

Biker888, 2018-12-15:
RGBS

There us a web UI ???

ezt írta:

ramapcsx2, 2018-12-16:
The web ui is the main control interface. You can find all the options there :)
https://github.com/ramapcsx2/gbs-control/wiki/Software-Setup#connecting-to-your-wifi

Biker888, 2018-12-22:
> @tracedebrake <https://github.com/tracedebrake>
> Hey, did you get any measurements or got any new insights?
> We already learned that the input AC coupling capacitors are of the wrong
> size, and that replacing them fixes dropouts caused by video level
> transitions.
>
> But there is at least one more thing:
> While large level swings don't cause issues anymore, a large amplitude
> (white screen) can still cause sync loss. This is what the extra
> termination resistor is intended to fix.
>
> I have a few devices that are still too "hot" though (newer PSX).
> The software can detect the dropout and lower the SOG slicer level (to
> about 3 out of 31).
> This works, but leaves little room for error and generally doesn't seem
> correct.
> I verified the peak to peak amplitude is 0.7Vpp at the SOG input pin,
> behind the AC coupling cap.
> The problem seems to be that the sync pulse is severely compressed from
> the white signal..

I had, have  no problems with the sync . The problem is that the output
frame with is narrow. So the board picks up the whole pic from the device
but the output misses some from the side. I can get it out by moving the H
pos  but then of course the other sde gets chopped off

ezt írta:

ramapcsx2, 2018-12-22:
Try another output resolution (preset). Try restarting your TV / display device (so it picks up the real size of the picture).

You only have to move positions if the display device doesn't do any size detection. Normally, this only happens on VGA > HDMI > Display setups.

If you can't get it right, use the horizontal scaling control to make the active video smaller.

Biker888, 2018-12-22:
The old device starts up after GBS does.

The output is a VGA 15"  LDC monitor.

Robert Neumann <notifications@github.com> ezt írta (időpont: 2018. dec.
22., Szo, 23:30):

Biker888, 2018-12-22:
So the LCD monitor has no issues... only the GBS uses a smaller frame. When
I adjust the HOR size on the GBS, i can not go "narrower" to see the whole
image width

Robert Neumann <notifications@github.com> ezt írta (időpont: 2018. dec.
22., Szo, 23:30):

ramapcsx2, 2018-12-23:
Okay, I understand the problem now. From Wikipedia:
"EGA is dual-sync; it scans at 21.8 kHz when 350-line modes are used and 15.7 kHz when 200-line modes are used."
I suppose this is using the 350 line mode, which my software isn't prepared to deal with.
It is possible to support EGA 350, but I don't have any devices that deliver this timing. 

In the web ui, there is a "print infos" button that prints to the text area on the bottom of the page.
Please enable this and paste a few lines here, so I can see what your device's timings are.

Biker888, 2018-12-23:
So the frame ends there on the righ side.. and starts again (so i can see
the same frame starting again).
That is the widest it goes (the GBD's frame)

my EGA (or CGA) signal uses a 25.5 KHz  / 60Hz signal
Here it is made "GBS compatible" by using the logical gates....
The WEB UI i have to discover... first how to access it..

[image: COMP sync.jpg]

Robert Neumann <notifications@github.com> ezt írta (időpont: 2018. dec.
23., V, 11:43):

ramapcsx2, 2018-12-26:
image: COMP sync.jpg doesn't show up, but I understand what you did (you built a sync combiner to get CSync).
If your TFT support the resolution, you can get better results by feeding H + V sync into the GBS.
Bascially, you need to undo the sync combiner and just send separated H + V Sync. The amplitude should be no more than 3.3Vpp !

I will add support for these EGA, or "mid-res" timings some day. Then they will work in upscaling modes as well.

Biker888, 2018-12-26:
Yep.
I made the XOR  + INV  gates to make a HV into a Csync (inverted) to work
with the GBS
The pic was too large to send (i have no direct EMIAL to you)

Here is my "devil" device H V  sync analised.....

So to even use the WEB UI with this GBS i will need an arduino board
attached to it...

Robert Neumann <notifications@github.com> ezt írta (időpont: 2018. dec.
26., Sze, 17:35):

Biker888, 2019-01-04:
So i guess whatever i do now... it will not help (needs some changes in the
FIRMWARE i guess)

..

Akos N. <mraq78@gmail.com> ezt írta (időpont: 2018. dec. 26., Sze, 20:50):

ramapcsx2, 2019-01-04:
@Biker888:
The software can handle your source, but you need to input separated HSync and VSync.
Remove your circuit that you built and change your cable so that HSync goes into the "S" pin and VSync goes into the "VS" pin ("VS" is marked on the board, right behind the VGA input connector).

You need an ESP8266 as controller, if you want to access the controls over WiFi.

Biker888, 2019-01-04:
Would it "fix" the issue ?
I can try.  Any changes with versions ? (some boards use different
firmware.. etc)

Robert Neumann <notifications@github.com> ezt írta (időpont: 2019. jan. 4.,
P, 18:38):

ramapcsx2, 2019-01-04:
It has a good chance of fixing the issue :)
The original firmware on the GBS does not matter, it is 100% bypassed anyway.

Biker888, 2019-01-04:
I mean that not using any additional MODs just this Sync "change"
I used  H V syncs like u suggested (just as it came out of the PC)... was
not working (so i can try....again)
(Maybe needs some inverting or something)

Robert Neumann <notifications@github.com> ezt írta (időpont: 2019. jan. 4.,
P, 18:55):

ramapcsx2, 2019-01-04:
Okay, then you need to get the web ui working.
https://github.com/ramapcsx2/gbs-control/wiki/Software-Setup#connecting-to-your-wifi

Biker888, 2019-01-04:
i connected the V H to all possible pins there...
WEB UI... so i need some ARDUINO stuff

Robert Neumann <notifications@github.com> ezt írta (időpont: 2019. jan. 4.,
P, 19:02):

closed 2019-03-09 by ramapcsx2