ramapcsx2/gbs-control #48: gbs-control:3109:35: error: 'HTML' was not declared in this scope

ngc20108, 2018-10-07:
Arduino: 1.8.7 (Windows 10), Board: "WeMos D1(Retired), 80 MHz, 115200, 4M (3M SPIFFS)"

C:\Users\RX\Downloads\gbs-control\gbs-control-master\gbs-control\gbs-control.ino: In function 'void handleRoot()':

gbs-control:3109:35: error: 'HTML' was not declared in this scope

   server.send_P(200, "text/html", HTML); // send_P removes the need for FPSTR(HTML)

                                   ^

Multiple libraries were found for "PersWiFiManager.h"
 Used: C:\Users\RX\Documents\Arduino\libraries\PersWiFiManager
 Not used: C:\Users\RX\Documents\Arduino\libraries\gbs-control-master
 Not used: D:\arduino-1.8.7\libraries\gbs-control
Multiple libraries were found for "ArduinoOTA.h"
 Used: C:\Users\RX\Documents\Arduino\libraries\ArduinoOTA
 Not used: C:\Users\RX\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\libraries\ArduinoOTA
 Not used: D:\arduino-1.8.7\libraries\gbs-control
exit status 1
'HTML' was not declared in this scope

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

ramapcsx2, 2018-10-08:
Current version of the ESP8266 core is 2.4.2. It may compile after you update.
If this version doesn't show up in the boards manager, try using the Arduino IDE beta (1.9.0).

ramapcsx2, 2018-10-15:
Is this still an issue?

ramapcsx2, 2018-10-20:
Probably fixed by user :p

closed 2018-10-20 by ramapcsx2