ramapcsx2/gbs-control #50: De-interlacing on amiga

ztxdk, 2018-10-15:
Hi not really an issue, but just tried this today with my v4 gbs, a wemos d1 mini and Amiga 1200. And wow the image is a lot better!
However I usually run workbench in the "Pal high res interlaced" and the stock gbs does an ok job of removing the flicker, except for a little flicker near the mouse pointer and a bit blurry text. But with gbs-control the image is super clear but flickering (interlaced), is there any way to get the crispness of gbs-control and the deinterlacing of the original?

ramapcsx2, 2018-10-16:
The stock firmware uses 2 buffers and the deinterlacing engine to blend them. You see this in the artefacts around moving objects.
My firmware ignores whether the content is interlaced or not. This allows "fake interlaced" 240p content to display properly, but it doesn't spend any resources on deinterlacing, so the image has interlace flicker.

At this point, I don't know how to set up a good looking deinterlace method.
It will require more memory for the buffers (and I can't tell if that memory is even available), and it will increase lag by 20ms for 50Hz PAL.
From previous experience, I expect a lot of research effort required to get it working at all.

There is another way:
If your display accepts the Amiga timings, then it may do a good deinterlacing job itself.
A test is simple, just try the Source Pass-through mode via web ui :)

ztxdk, 2018-10-17:
Ah ok, thanks for the explanation!
And no, my monitor doesn't take the amiga timings..
But another question, please redirect me if there is a better place for it!

Since I completely forgot that the Amiga workbench is able to run in vga, I've made a vga cable for the amiga, with a 74HCT14 inverter. 
It gives a decent picture with workbench in vga screenmodes, and I can connect that cable to the gbs  vga-in and it passes the image thru.
This cable uses the h/v sync lines, but the old cable (from the amiga to the gbs) uses the c sync line (to the connector just behind the vga-in on the gbs). If I change to native mode on the amiga and disconnect h/v sync and connect the c sync, it (the gbs) show the converted picture.
If I dont switch the sync signals the image is just passed thru to the monitor and it cannot show it (shows out of range and says it's in 15 khz).

Is it possible to do this automatic (in software on the gbs/esp without the c sync signal) or do I need a switch for the sync signals. As far as I can tell the vga-in and the connector right behind it on the gbs are connected.

Regards,
ztxdk (martin)

ramapcsx2, 2018-10-17:
VGA mode is a nice solution to the problem. Anything that avoids interlacing altogether will look better.

I'm not sure I understand your question though.
Do you have 2 cables connected at the same time, and both cables output a signal?
If so, the "S" pin and the HSync pin are shared, so having a signal on both will be bad.

ztxdk, 2018-10-17:
I only have one cable connected at a time..
The normal cable I use is RGB and C sync.  

Right now I have a amiga to vga adapter on a breadboard, if I switch to a native amiga screenmode and pull the H and V sync and connect C sync to the gbs where H sync was, it (the gbs) converts the native resolution..

ramapcsx2, 2018-10-17:
Right, the software goes through 3 possible input modes and looks for valid sync on them.
- RGBS on logical input #1: CSync (or CV / Luma) need to appear on the SOG1 input > Scaler goes into normal (upscaling) operation
- RGBHV on logical input #1: VSync needs to appear on VSin > Scaler goes into RGBHV bypass
- YPbPr on  logical input #0 (RCA plugs): CSync needs to appear on the SOG0 input > Scaler goes into normal (upscaling) operation

So for your case, it can either detect VSync for RGBHV bypass, or detect CSync for normal upscaling.
The software so far cannot do normal upscaling mode with RGBHV, but it can do a bypass mode with CSync.

ramapcsx2, 2018-10-26:
If you're still hoping for deinterlacing support, keep watching the development branch. I think I've got something soon :)

ztxdk, 2018-10-29:
Awesome, cool, will check it out asap!

ztxdk, 2018-10-31:
Wow this is good! As in REALLY good! The image in the "Pal: High Res Laced" resolution on the Amiga (640x532) and output in 1280x720 looks so good. Even better than native amiga vga on my monitor (it doesn't handle 4:3 very well)... The only flickering I've noticed is just at the edges of the mouse pointer or if something big is being moved around on the screen really fast, and this is not a problem for me.

The only issue is some flickering/blanking once in a while (where the image goes black for less than half a second), but this disappears if I pull the wire from the debug pin.

ramapcsx2, 2018-11-01:
Hmm, check the serial output when it blanks to see if it prints something.
If it's just a dot (.), then it means it was a momentary sync loss.

The quality is so-so for moving content. I'll see what improvements I can get out of it yet :)

ramapcsx2, 2018-12-22:
The deinterlacer has been implemented and looks pretty good, imho. Closing.

closed 2018-12-22 by ramapcsx2

lovrenco, 2021-01-30:
Deinterlacing works good on static screen, bit not on mouse pointer scrolling over icons on the desktop on Amiga1200, resolution Pal high res interlaced. 
I discovered that you can reduce artefacts by using MagicTV and 16 color mode (rebel optimized palette). 
However, as I once had external amiga scandoubler, I can see the difference. 
Unit works OK with non interlaced resolutions, but I would like to have higher resolution option for A1200...