ramapcsx2/gbs-control #53: Output Format: VGA / Component Video

tvirusx, 2018-12-11:
hi there... is this feature "Output Format: VGA / Component Video" working, if so what would it be the pin out to build a VGA to a component cable. I found this, but now sure if it will work:

[image]

Thanks once again

tvirusx, 2018-12-12:
Found it... u kept the same rgb r for red g for green y cable and b for blue... thanks this is the greatest thing ever kkkk I can use it in all TVs now =)
u guys are awesome

ramapcsx2, 2018-12-15:
Cheers :)
I think every adapter or adapter cable for VGA to Component is using the same pinout (R > Pr, B > Pb, G > Y).
It wouldn't make any sense to have it different :)

closed 2018-12-15 by ramapcsx2

partlyhuman, 2021-11-17:
Hmmm. I'd love to figure out what I'm doing wrong. I'm downscaling to 15khz and my R-Y channel seems to be missing. Image renders blues, greens, yellows, no red; and removing the Pr cable from the TV doesn't change the image. HOWEVER,
I've used both a HD15-Component breakout cable I had onhand, as well as manually wiring one following the same diagram using a VGA breakout box. In both cases, I'm still missing the R channel.

Thinking it's an input problem, I've tried both Raspi 3 using 480p HDMI→VGA and 480p PS3 with AV component cable out, and both exhibited missing reds.

My GBS-C has no issues displaying red when upscaling and in VGA output mode.

I wonder if there's anything I can do to debug this!

partlyhuman, 2021-11-17:
Dangit.... it is happening while upscaling too. I thought it wasn't.

I detected a short between R and GND so it's on me to figure this out. Thanks!