ramapcsx2/gbs-control #54: I can't compile the code to my Arduino Uno

Vininess, 2018-12-20:
I tried to compile the latest code, but i always get this error "'saveUserPrefs' was not declared in this scope"

Thanks

ramapcsx2, 2018-12-22:
Hey,
I stopped supporting AVR based Arduinos some time ago.
You'll need an ESP8266 based device now.

Basically, the project outgrew the small Arduinos and I had to step up to ESP8266 at least.

Did you read the Uno recommendation somewhere? If so, I need to update those docs ;)

Vininess, 2018-12-22:
I see
"Did you read the Uno recommendation somewhere?"
No no, I read all the articles about your code and i will buy the ESP8266, because of the features
And i know all the limitations of the Arduino UNO

But my GBS8200 got some strange behaviours
I bought second hand and i think that have some bug in the firmware

Frequently gets "No Signal" on my monitor or buttons don't work
Because of that i want to install the code on Arduino UNO to do some tests and see if is a firmware bug

Can you help me? just to see if that board works or not

Really thanks for the help

ramapcsx2, 2018-12-22:
That's just how these boards ship. The firmware on them is not optimized for classic consoles. Some sources will work, others have no chance with the botched SOG settings.

Instead of trying to get the Uno to work, try using a different console.
A PS2 using Component cables will work, for example.

But anyway, here is an older build that you can use on your Uno:
https://github.com/ramapcsx2/gbs-control/tree/167cb1fe00aadfb44bd5442d2d6dc86b738d79b8
(Use the "Clone or download" button)

Vininess, 2018-12-22:
I tried using a Wii with Component cables
But some times works, some times works for a while, some times works but the menu don't work

I coded the arduino, but don't work
Wired PIN 10 to Degub
PIN A5 to SCL
PIN A4 to SDA
Jumped the P8

I connected the Wii via Component Cable, and shows nothing

Something i miss?

ramapcsx2, 2018-12-22:
If you suspect a wiring issue, check the pictures and description on
https://github.com/ramapcsx2/gbs-control/wiki/Build-the-Hardware
On an Uno, you need 3 data wires (SDA, SCL, debug) and 1 power wire (just Ground).
The Uno itself gets power either via USB, or any other method you normally use.
(Note: I don't recommend using the GBS board 3.3V to power the Uno anymore.)

I can't offer any more low level support, especially from afar. 
The Wiki guide is all you would need with an ESP8266.

Vininess, 2018-12-26:
Really Thanks for the help

It's work, but this GBS really get some problems
I'd sent it back, bought another GBS and bought an ESP8266
And i will try the full setup

Again
Thanks

goutsune, 2019-01-02:
I've also experienced some sync issues with GBS and Wii, one thing I've noticed is that it overscans picture a lot. Another thing is that when screen gets bright (e.g. when you open a channel) GBS locks sync for a few frames. I don't have oscilloscope to provide more detail here but I can do tests if @ramapcsx2 might want that.

P.S. I work round the issue by using bypass mode in gbs-control, most games support 480p anyway.

ramapcsx2, 2019-01-02:
See here for the sync issues: https://github.com/ramapcsx2/gbs-control/wiki/Sync-on-Green-Capacitor-Replacements

Overscan happens when the source draws more active video than what older consoles used to do.
I optimized for older designs, so some manual adjustment is necessary with new consoles.

ramapcsx2, 2019-01-07:
Marking as resolved.

closed 2019-01-07 by ramapcsx2