ramapcsx2/gbs-control #56: Sacing slots no working on new update

tvirusx, 2019-01-03:
Hi there...
Just did an update and try to save the settings but when I do select the slot it gives me: 

saveUserPrefs: open file failure

and if I try to save it says that I need to select a slot first.

thanks once again

Thompson

ramapcsx2, 2019-01-04:
Did you update your Arduino environment lately?
Check the build options and make sure that "Flash Size: 4M (1M SPIFFS)" is selected.

Somehow the defaults changed to having SPIFFS disabled.
If that's intended, I need to add some more checks..

ramapcsx2, 2019-01-05:
I've added a screenshot for this to the Wiki :)
https://github.com/ramapcsx2/gbs-control/wiki/Software-Setup#esp8266-ide-settings

tvirusx, 2019-01-06:
Yep... that did the trick.... thanks very very much u guys awesome...😆😀

closed 2019-01-06 by tvirusx