ramapcsx2/gbs-control #60: The background is filled with green from version 28.01.2019; image is cut horizontally

nimamov, 2019-02-07:
Hello rama!

I use GBS-8220 (RGBS2VGA) with your firmware. I connect the adapter to two different Soviet computers:

1. Computer "Soyuz-Neon PK-11/16", which uses 300 visible lines in a video frame (50Hz, 15.6259kHz) and this does not fit well with television standards. As a consequence, the GBS-8220 with factory firmware shows an image, but it twitches. GBS-8220 with your firmware does not detect the synchronization signal at all, what is reported in the logs.
Is it possible to fix something in the firmware to capture synchronization from this computer? The framemeister does this without problems.

2. Computer UKNC (MS0511), the resolution of the video frame 640 * 288. There are two problems. First: horizontally, the entire frame can fit into the screen only in adapter mode 640 * 480. In other modes, no settings help, the frame goes left or right. 640 * 480 is also a good result, but in this mode the image is blurred in some places. Is there any way to fix this for all modes? The second problem is that since the firmware version 28.01.2019, the background of the image of my UKSC has become permanently green. I managed to fix it by replacing in the latest firmware version in the line 4818 in file gbs-control.ino:
4818: updateClampPosition (1);
I replace unit to zero, and then the background turned black again. Can this be taken into account in further development?

ramapcsx2, 2019-02-07:
Hi,
these are some interesting computers!
The Союз-Неон is probably older than most things that output to a TV, right?
The МС 0511 appears to be recent enough to output real PAL timings.

For the Союз, I can only guess what happens. It probably is not being recognized by my programmed timings. It is neither NTSC, nor PAL, but it is low resolution.
You can try to "disable syncwatcher" via the web interface. Then manually load any of the presets and see what you get.
Please post a log of the gbscontrol output. Maybe I can guess what's wrong.

For the МС 0511: 
UpdateClampPosition() adjusts the black level clamping point.
It can fail if the video at the determined position is not actually black, and this may be the issue here.
Again, I would need to have a test example to fix this.
Until I figure it out, please just disable that line of code (updateClampPosition (1)).
I assume other old home computers may have similar problems. I just need to find one.

You can use the web interface to correct the image position.

nimamov, 2019-02-07:
Союз-Неон and МС0511 are about the same age, 1989-1993. They are exactly the same as the BK0011M can connect to SCART, but on most TVs the frame does not fit completely within the limits of the screen.
I tried absolutely all the presets for your firmware for Soyuz-Neon, but in each case I saw a message in the log file that no synchronization was detected.

About МС0511 You did not answer, is it possible to do something with the fact that in all modes (except 640 * 480) the image does not fit horizontally on the screen?

Also, synchronization with Союз-Неон and МС0511 easily captures the TV tuner Behold TV H8, which has an RGB + S input. Here are screenshots from Behold TV H8:
МС0511:
  https://cdn.freelance.ru/download/1073874/%C1%E5%E7%FB%EC%FF%ED%ED%FB%E9.jpg
  
Союз-Неон:
  https://cdn.freelance.ru/download/1073875/%C1%E5%E7%FB%EC%FF%ED%ED%FF%%91.jpg
  
  https://www.youtube.com/watch?v=MkiF7XuDPmE

ramapcsx2, 2019-02-08:
The Youtube video shows a rolling screen, meaning gbscontrol doesn't see a valid VSync signal. There are many different causes for that. I need to see some logs.

Regarding the image position, the source probably has a large overscan area (seeing as the clamp position update fails as well).
Just try to center the picture, while avoiding image artefacts.
You can scale horizontally, vertically, move the inner canvas in all 4 directions and even move the HSync position.
There is a way to make it fit, even in the higher resolution presets.

nimamov, 2019-02-08:
Sorry, YouTube video does not apply to gbs-8220, this is capture by the Behold tuner. I published it to confirm that standard RGB+S capture devices understand the signal from Soyuz-Neon.

gbs-8220 with your firmware writes in the logs that the Soyuz-neon synchronization signal was not detected:
  no sync found
  input detect mode
  <reset>
  found: 5 getVideoMode: 0 input: 1
  found: 5 getVideoMode: 0 input: 1
  Auto SOG: retry #1
  Auto SOG: retry #2
  Auto SOG: retry #3
  SOG level: 1
There is no other information in the logs.
gbs-8220 with factory firmware signal detects, but the image is twitching.
--------------------------------------------------------------------------------
MS0511:
>> You can scale horizontally, vertically, move the inner canvas in all 4 directions and even move the HSync position.
I tried these settings very carefully and they help in 640x480 mode. In other modes, change HSync position could help, but after a certain position the sync signal is lost and the image disappears. If return HSync position - the image appears again.

ramapcsx2, 2019-02-08:
"getVideoMode: 0" << Yeah, the timing is non-standard somehow. I can't fix this unless I have a similar system.

Can you take a picture of what 1280x960 looks like by default?

nimamov, 2019-02-09:
I took video manipulations with Scaling Horisontal +/- and Pictire right/left in 1920x1080 mode. In 1280x960 mode, the behavior is similar.
The green borders on the right and left, I deliberately exposed, to show the obviousness of the problem.
[https://www.youtube.com/watch?v=M4AZryrbDQM](url)

nimamov, 2019-02-09:
On the problem of the lack of a sync signal Soyuz-Neon, my friend provided you with a simulated sync signal and its parameters: https://ibb.co/61b9wDw
Will this help fix the firmware gbs-8220 for Soyuz-Neon?

ramapcsx2, 2019-02-09:
The sync drawing looks okay to me. VSync is present in the CSync signal.
You are using CSync ("Sync" in the drawing), right? 
"Hide" is just the blanking signal and is not used for analog display.
If you accidentality used the "Hide" signal, that would explain why there is no sync detected.

The Youtube video shows that everything is as I designed it (sorry :p).
I optimized the presets for video game consoles, which all have much smaller overdraw than home computers.
As a result, a few pixels will be cut off, if the source uses a lot of overscan.
You should adjust the display so that the left side shows as much text as possible.
Then increase the horizontal mask until all of the green area and any other issues are hidden.
This is the best it will get for the HD presets.

nimamov, 2019-02-10:
_>> You are using CSync ("Sync" in the drawing), right?_
Yes, this is a comprehensive video sync signal.

_If you accidentality used the "Hide" signal, that would explain why there is no sync detected._
No, "Hide" is only a logical illustration of the moment of extinguishing frame impulse and line impulse. "Hide" does not physically exist.

_>> The Youtube video shows that everything is as I designed it_ 
Please look again carefully video. When I fully extend the right edge of the image from the green field, the left edge is cut off by almost on one sign-place. And vice versa. This can not be eliminated by any settings. It helps only the inclusion of 640x480 mode.

_>> Я оптимизировал пресеты для игровых приставок,_
Please help us optimize the gbs-8220 for these two computers, and then the gbs-8220 will be quite better than framemeister!

_>> As a result, a few pixels will be cut off, if the source uses a lot of overscan._
Is it really impossible to do anything with this problem with the help of changes in the firmware? Why is there no such problem in 640x480 mode?
How did the framemeister developers solve this problem?

ramapcsx2, 2019-02-10:
The problem can be easily solved by dropping quality. I don't want to drop the quality though :)

Well, maybe I'll add something like a compatibility mode, just for sources like these home computers.

nimamov, 2019-02-10:
_>> Well, maybe I'll add something like a compatibility mode, just for sources like these home computers._
Yes, that would be wonderful. Let's hope and wait. And I think that if you add a mode for these two computers, this will cover a very large assortment of Western and Russian personal retro computers.
Thousands more people will be grateful to you, especially in Russia.

_>> Gbscontrol is just me, developing something as a hobby :)_
The BK0011M with your firmware operates at a frequency of 48.8 Hz (in 1920x1080 mode), which even the framemeister is not able to. You have already surpassed this development team alone and we believe that you are capable of more!

closed 2019-03-09 by ramapcsx2