ramapcsx2/gbs-control #61: Possibility of creation 2560x480/2560x512 super resolutions for PSX/PS2

goutsune, 2019-02-24:
Hello again, I've been wondering if it is possible for GBS boards to clock pixels at such resolution so we can get seamless switch between 512/640px sources as these seem to get switched a lot while in game. @ramapcsx2 if you can point me at what registers to tinker with in 640px presets I'd be more than happy to try that myself :^)

ramapcsx2, 2019-02-25:
It is much more complex than that ;p

Sampling starts with the ADC sampling clock (which already is at 2560px, btw). The acquired data gets parsed by the Input Formatter, which applies some restrictions, such as a 1024 px maximum line length.
After all kinds of processing on that maximum of 1024 px, the data gets scaled onto the output resolution.

So your idea is already implemented as best as possible. 
I try to maximize the sampling rates and optimize the usage of the digital line buffers. 
In the final step, I oversample the display clock as well.
It seems to matter most to have a high sampling rate and to then not loose definition by the output scaling, and this is how the default presets are setup currently.

goutsune, 2019-02-25:
Hmm, it makes me wonder then why I can get pixel perfect 640px resolution just ok, but have to squish image into square to get 512px pixel perfect image. Are you sure feedback presets do that too?

ramapcsx2, 2019-02-25:
I don't understand what exactly you're describing. I can tell you however that the feedback clock preset isn't working well. The idea of it was to output something similar to the input, but it doesn't appear to work well for many source resolutions.

The other presets are all converted to use the 2560px ADC sampling clock > data reduction > digital processing > upscaling to about 2000px wide times hotizontal lines of the preset, fit into the timings of the preset.
The last part of the chain is actually the "super resolution" part. One could take all the parameters and try to calculate how many "pixels" wide it is, but this is beyond me. Too many parameters are hidden anyway. All I can say is that I maximize it, then hope for the best (and it works well here :)).

ramapcsx2, 2019-04-04:
Closing as the general idea already is a design goal of the project.

closed 2019-04-04 by ramapcsx2