ramapcsx2/gbs-control #62: No picture

milika, 2019-03-26:
Hi,
connected and reconnected few times, changed esp8266 and still same problem.

Using amiga via composite pal and cant get any picture, at all. Without amiga, with composite, with rgb, without, no picture.

This is the output when composite is connected:
`found: 138 getVideoMode: 0 input: 1
VS active
HS active
RGBHV bypass 
RGBHV: HS active low
post preset done (preset id: EC) (custom)
!!!!!!!!!!<reset>`

ramapcsx2, 2019-03-29:
Hey,
not all details of your setup are clear to me, but I can see that the scaler identifies an RGB+HSync/VSync signal.
Is this what you intended to do? Does your cable carry HSync and VSync?

milika, 2019-03-29:
I tried composite (Y) only, and RGBS (combined sync) same message.
Could my unit be faulty? Should there be any picture without any source?

ramapcsx2, 2019-03-29:
Try the composite (Y) only cable into the GBS green Component input. (It is an RCA "yellow plug" cable, right?)
This should definitely give you a picture with an Amiga and confirm that it's working at least / as a starting point.
The picture will be black and white :p

milika, 2019-03-29:
Tried exactly that - still the same :)

ramapcsx2, 2019-03-29:
What is the log output then? Oh, and please update the software. I just pushed a fix to a bug that I introduced some time ago.

milika, 2019-03-29:
Ill try new fix.
exactly the same log :) whatever i do

ramapcsx2, 2019-03-29:
So you get "VS active HS active RGBHV bypass" even on the Component input? Are all other video connections from the Amiga disconnected?

If so, then there is an installation issue and the software on the ESP8266 reads garbage data.

milika, 2019-03-29:
Ok, new source uploaded, connected old DVD player via YUV now now get this:
`*!**!**!*<reset>
Activity detected, input: Component
 YUV
.*!*!*<reset>
Activity detected, input: Component
 YUV
*!***!*<reset>`

milika, 2019-03-29:
Video timings:
`*!**!*
htotal: 2150
HS ST/SP     : 2159 2236
HB ST/SP(dis): 2175 2236
HB ST/SP     : 2191 2236
----
vtotal: 138
VS ST/SP     : 173 188
VB ST/SP(dis): 189 188
VB ST/SP     : 720 188
IF_VB_ST/SP  : 720 965
*!**!*<reset>`

milika, 2019-03-29:
and options:

preset preference = 1
frame time lock = 1
preset slot = 1
frame lock method = 1
auto gain = 0
scanlines = 0
component output = 0
deinterlacer mode = 1
line filter = 0
peaking = 0
pal force60 = 1
/userprefs.txt 11
preset preference = 1
frame time lock = 1
preset slot = 1
frame lock method = 1
auto gain = 0
scanlines = 0
component output = 0
deinterlacer mode = 1
line filter = 0

ramapcsx2, 2019-03-29:
Please try clearing all settings to default.
In the Arduino IDE, select Tools > Erase Flash > All Flash Contents, then upload the sketch once.
Turn Erase Flash back to Only Sketch afterwards.
You'll have to do the WiFi setup again, so beware of that.

When you test the scaler after clearing the settings, please don't adjust any of the settings.
It has to work out of the box, and if it doesn't, there is some kind of issue to fix first.

milika, 2019-03-29:
did it, all reset and everything and again:
*!**!**!**!*<reset>
Activity detected, input: Component
 YUV
.*!**!**!**!**!*<reset>
Activity detected, input: Component
 YUV
.*!**!**!**!**!*<reset>
Activity detected, input: Component
 YUV
.*!**!**!**

ramapcsx2, 2019-03-29:
Does (did) your GBS work without the ESP8266?

milika, 2019-03-29:
Yup,  but awful, got menus but could not sync with amiga. Did not try via VHS, do not know if it still works. Would need to desolder...

ramapcsx2, 2019-03-29:
Okay, I think there is a wiring problem or similar. 
There are some weird values in some of your logs, such as "vtotal: 138" (that's totally off the mark) or "preset id: EC" (I don't have such an ID).

Other values appear to be valid though, so maybe the fault is intermittent.

ramapcsx2, 2019-03-29:
Is the jumper to disable the onboard MCU installed?

milika, 2019-03-29:
yes,im trying to send images now 
I resoldered it 3 times, tried switching sdl/sda (did not dettect ic then) used 2 d1 minis

milika, 2019-03-29:
https://ibb.co/g9kSGdG
https://ibb.co/Kz84HHw
https://ibb.co/tCBv1Tp
https://ibb.co/ZmHnRLC

ramapcsx2, 2019-03-29:
Okay, I can't see an obvious problem in your pictures.
Please get one more cable for another ground connection. Use the free GND pin on the GBS and find another ground point on the ESP8266 board (if there aren't any more, re-use the one you use now).
The additional wire will make the communication more reliable.

If that doesn't fix it, please have the auto detect run for a minute or two, then copy the entire log and paste it somewhere.
I can read some more stuff from a complete log.

milika, 2019-03-29:
how do i get the complete log?

ramapcsx2, 2019-03-29:
By having a terminal open with the computer connected to the ESP8266 (as with sketch uploads). In the Arduino IDE: Tools > Serial Monitor
Set the baud rate to 115200 if you only get garbage symbols.

milika, 2019-03-29:
Done, massive wire ground 2mm added
log:
starting
(WiFi) AP mode (SSID: gbscontrol, pass 'qqqqqqqq'): Access 'gbscontrol.local' in your browser
userprefs open ok
preset preference = 0
frame time lock = 0
preset slot = 1
frame lock method = 0
auto gain = 0
scanlines = 0
component output = 0
deinterlacer mode = 0
line filter = 1
peaking = 1
pal force 60 = 0
i2c: 0
<reset>
Activity detected, input: Component
 YUV
.*!**!**!**!**!*<reset>
Activity detected, input: Component
 YUV
.*!**!**

ramapcsx2, 2019-03-29:
Do you still have the PC connected / terminal open?
Try sending the letter "e" (no quotes, just e), then send "d" and paste the log.
Also you can try the information readout mode by sending "i". A short (20 line) log of that would also help.

I have a hunch there will be bad data in the "d"ump ;)

milika, 2019-03-29:
here you go: :)

*!**!*<reset>
Activity detected, input: Component
 YUV
.*post preset done (preset id: D0) (custom)
!**!**!*0x0, // s0_40
0x0, // s0_41
0x0, // s0_42
0x0, // s0_43
0x0, // s0_44
0x0, // s0_45
0x0, // s0_46
0x0, // s0_47
0x0, // s0_48
0x0, // s0_49
0x0, // s0_4A
0x0, // s0_4B
0x0, // s0_4C
0x0, // s0_4D
0x0, // s0_4E
0x0, // s0_4F
0x0, // s0_50
0x0, // s0_51
0x0, // s0_52
0x0, // s0_53
0x0, // s0_54
0x0, // s0_55
0x0, // s0_56
0x0, // s0_57
0x0, // s0_58
0x0, // s0_59
0x0, // s0_5A
0x0, // s0_5B
0x0, // s0_5C
0x0, // s0_5D
0x0, // s0_5E
0x0, // s0_5F
0x0, // s0_90
0x0, // s0_91
0x0, // s0_92
0x0, // s0_93
0x0, // s0_94
0x0, // s0_95
0x0, // s0_96
0x0, // s0_97
0x0, // s0_98
0x0, // s0_99
0x0, // s0_9A
0x0, // s0_9B
0x0, // s0_9C
0x0, // s0_9D
0x0, // s0_9E
0x0, // s0_9F
0x1, // s1_0
0x1, // s1_1
0x1, // s1_2
0x1, // s1_3
0x1, // s1_4
0x1, // s1_5
0x1, // s1_6
0x1, // s1_7
0x1, // s1_8
0x1, // s1_9
0x1, // s1_A
0x1, // s1_B
0x1, // s1_C
0x1, // s1_D
0x1, // s1_E
0x1, // s1_F
[... 615 lines cut ...]
h: 120 v:  60 PLLx00 A:383838 S:38.9c I:9c D:ed9c m:0 ht:  38 vt: 38 hpw:3484 s: 0 W:31 F:37400 L:4
h: 120 v:  60 PLLx00 A:5d5d5d S:5d.9c I:9c D:ed9c m:0 ht:  38 vt: 38 hpw:3484 s: 0 W:31 F:37400 L:3
h: 120 v:  60 PLLx00 A:bababa S:ba.9c I:9c D:ed9c m:0 ht:  38 vt: 38 hpw:3484 s: 0 W:31 F:37400 L:3
h: 120 v:  60 PLLx00 A:ababab S:ab.9c I:9c D:ed9c m:0 ht:  38 vt: 38 hpw:3484 s: 0 W:31 F:37400 L:4
h: 120 v:  60 PLLx00 A:c1c1c1 S:c1.9c I:9c D:ed9c m:0 ht:  38 vt: 38 hpw:3484 s: 0 W:31 F:37400 L:3
h: 120 v:  60 PLLx00 A:e1e1e1 S:e1.9c I:9c D:ed9c m:0 ht:  38 vt: 38 hpw:3484 s: 0 W:31 F:37400 L:4
h: 120 v:  60 PLLx00 A:fdfdfd S:fd.9c I:9c D:ed9c m:0 ht:  38 vt: 38 hpw:3484 s: 0 W:31 F:37400 L:4
h: 120 v:  60 PLLx00 A:252525 S:25.9c I:9c D:ed9c m:0 ht:  38 vt: 38 hpw:3484 s: 0 W:31 F:37400 L:3
h: 120 v:  60 PLLx00 A:787878 S:78.9c I:9c D:ed9c m:0 ht:  38 vt: 38 hpw:3484 s: 0 W:31 F:37400 L:3
h: 120 v:  60 PLLx00 A:646464 S:64.9c I:9c D:ed9c m:0 ht:  38 vt: 38 hpw:3484 s: 0 W:31 F:37400 L:4
h: 120 v:  60 PLLx00 A:8a8a8a S:8a.9c I:9c D:ed9c m:0 ht:  38 vt: 38 hpw:3484 s: 0 W:31 F:37400 L:3
h: 120 v:  60 PLLx00 A:bebebe S:be.9c I:9c D:ed9c m:0 ht:  38 vt: 38 hpw:3484 s: 0 W:31 F:37400 L:3
h: 120 v:  60 PLLx00 A:c6c6c6 S:c6.9c I:9c D:ed9c m:0 ht:  38 vt: 38 hpw:3484 s: 0 W:31 F:37400 L:4
h: 120 v:  60 PLLx00 A:eeeeee S:ee.9c I:9c D:ed9c m:0 ht:  38 vt: 38 hpw:3484 s: 0 W:31 F:37400 L:4
h: 120 v:  60 PLLx00 A:5a5a5a S:5a.9c I:9c D:ed9c m:0 ht:  38 vt: 38 hpw:3484 s: 0 W:31 F:37400 L:4
h: 120 v:  60 PLLx00 A:323232 S:32.9c I:9c D:ed9c m:0 ht:  38 vt: 38 hpw:3484 s: 0 W:31 F:37400 L:4
h: 120 v:  60 PLLx00 A:535353 S:53.9c I:9c D:ed9c m:0 ht:  38 vt: 38 hpw:3484 s: 0 W:31 F:37400 L:3
h: 120 v:  60 PLLx00 A:7d7d7d S:7d.9c I:9c D:ed9c m:0 ht:  38 vt: 38 hpw:3484 s: 0 W:31 F:37400 L:4
h: 120 v:  60 PLLx00 A:8a8a8a S:8a.9c I:9c D:ed9c m:0 ht:  38 vt: 38 hpw:3484 s: 0 W:31 F:37400 L:4
h: 120 v:  60 PLLx00 A:b7b7b7 S:b7.9c I:9c D:ed9c m:0 ht:  38 vt: 38 hpw:3484 s: 0 W:31 F:37400 L:3
h: 120 v:  60 PLLx00 A:e2e2e2 S:e2.9c I:9c D:ed9c m:0 ht:  38 vt: 38 hpw:3484 s: 0 W:31 F:37400 L:3
h: 120 v:  60 PLLx00 A:f1f1f1 S:f1.9c I:9c D:ed9c m:0 ht:  38 vt: 38 hpw:3484 s: 0 W:31 F:37400 L:4
h: 120 v:  60 PLLx00 A:1c1c1c S:1c.9c I:9c D:ed9c m:0 ht:  38 vt: 38 hpw:3484 s: 0 W:31 F:37400 L:3
h: 120 v:  60 PLLx00 A:454545 S:45.9c I:9c D:ed9c m:0 ht:  38 vt: 38 hpw:3484 s: 0 W:31 F:37400 L:3
!**!*

ramapcsx2, 2019-03-29:
Yeah, the entire dump is uninitialized. 
The I2C comms must be bad, nothing is being written to the scaler chip and the readings in info mode are random bits.
Possibly the I2C pins on the ESP8266 are wrong..

milika, 2019-03-29:
on D1 mini SDA and SCL pins are d1 and d2
https://escapequotes.net/wp-content/uploads/2016/02/esp8266-wemos-d1-mini-pinout.png

milika, 2019-03-29:
D1 - goes to scl, D2 goes to sdl on board, i think that should be fine
as per https://ibb.co/tCBv1Tp

ramapcsx2, 2019-03-29:
Yep, that looks correct. The fault remains that no data is being written to the scaler chip.
You can test if a write is successful by doing this:
first, send "m" to disable the auto sync detection (syncwatcher)
next, send "s2s00s33" (it should return: S2R0x0 (was 0xFF) is now: 0x33)
then read it back by sending "g2g00" (it should return G2R0x0 value: 0x33)

You can use these steps to debug the wiring problem (I assume that's what it is).

milika, 2019-03-29:
s2s00s33 returns
S2R0x0 (was 0x0) is now: 0x0
not now: 0x33
- it means bad wiring?

milika, 2019-03-29:
G2R0x0 value: 0x0
also

ramapcsx2, 2019-03-29:
Yes, the values the ESP8266 wants to write don't get through.
I'm thinking about it..
It could be an Arduino ESP8266 core pin assignment issue, so that SCL and SDA aren't actually what the hardware uses for I2C.
Do you have a multimeter nearby?

milika, 2019-03-29:
sure, and basic oscilloscope if needed

ramapcsx2, 2019-03-29:
Here someone with a Wemos D1 Mini had similar issues and got them fixed: https://github.com/ramapcsx2/gbs-control/issues/47#issuecomment-435677165

ramapcsx2, 2019-03-29:
Oh, a scope! Awesome.

Can you see activity on your D1 and D2 that looks like I2C? :)

milika, 2019-03-29:
https://ibb.co/PWw5KFQ - scl 
https://ibb.co/yWYs2gH - sda

milika, 2019-03-29:
hope it helps

milika, 2019-03-29:
hm i see we need lower time space
https://ibb.co/KVk4bCK - sda
it looks like its trying

ramapcsx2, 2019-03-29:
Hmm, there is some activity that could be I2C alright. The default state is high and the ESP does seem to pull it low to communicate.

Let's focus on the I2C only.
Here is a sketch that you can flash onto your ESP. It runs an I2C address check and should find 2 devices on the bus, if I2C works correctly:
http://www.esp8266learning.com/i2c-scanner.php

ramapcsx2, 2019-03-29:
Setting the baud rate to 9600 in the monitor and resetting the ESP, I get the following:
$nٴ⸮ dr[⸮h⸮⸮>I⸮
I2C Scanner
Scanning...
I2C device found at address 0x17  !
I2C device found at address 0x50  !
done

And that is correct.
(One device is the scaler chip, the other is a small EEPROM on the same bus.)

milika, 2019-03-29:
Scanning...
I2C device found at address 0x50  !
done

only this, so communication is the problem. 
Should i reverse sda scl? or... :)

ramapcsx2, 2019-03-29:
Now that's super weird.. The 2 devices should be on the same bus, and you say the GBS worked (kind of) with the original MCU.
I sure hope they didn't change the hardware design or something..

milika, 2019-03-29:
so which device is missing, perhaps i can check its input voltage at least?

milika, 2019-03-29:
I see its rom chip

ramapcsx2, 2019-03-29:
The scaler at 0x17 is missing. The EEPROM at 0x50 responds.
If you can do a continuity check, there are 2 resistors marked R36 and R37, right next to the original MCU.
The side of the resistors facing the MCU should have continuity with the small 8 pin chip nearby, on pins 5 and 6 of it. This is the I2C bus.
Since the sketch can see the EEPROM, I assume there is continuity.

Once that is established, we'll go from there.

milika, 2019-03-29:
Both of them have continuity with top left pin of small 8pin chip marked IC8

ramapcsx2, 2019-03-29:
The other side of the resistors then, please. (You just found Vcc :) )

milika, 2019-03-29:
other side have some resistance to pin 2,7k r36 and 1.3k r37 i think, but definitely no brakage

milika, 2019-03-29:
and voltage on that pin is 3.35v

ramapcsx2, 2019-03-29:
The GBS needs to be off!

milika, 2019-03-29:
I tested with off :) just tried voltage - on then tried, now off again :)

ramapcsx2, 2019-03-29:
Ok, the important check, please see this professional picture:
https://i.imgur.com/4pKrSQg.png

Basicially, test for continuity between those 2 resistors and the 2 outermost pins on the scaler chip.
We just want to know if both resistors are attached to one of the scaler pins (the order isn't too important).

milika, 2019-03-29:
continuity is present near c50 cand c30

ramapcsx2, 2019-03-29:
That location sounds more like ground? 
It has to be just one resistor and one of the 2 outermost pins on the scaler, each.

milika, 2019-03-29:
R10 and R11 have continuity to 1st 2 pins, sorry checked from R36 and R37

ramapcsx2, 2019-03-30:
Sorry, I didn't get your latest update 40 minutes ago..

So okay, the electrical connection is there so it should work.
If you measure resistance between your ESP D1 / D2 and those scaler IC pins, you should get around 33 Ohms (which is the value of the 2 dampening resistors).

milika, 2019-03-30:
33ohms on both pins :)

ramapcsx2, 2019-03-30:
So then, why doesn't the scaler IC reply on its address.. Any idea?
If you can't get any more hints, you can try desoldering Vcc and Gnd from the ESP board, taking the jumper off, then powering the GBS.
It should work on the onboard MCU then.
You should see a "no signal" message that is powered from the onboard MCU OSD generator.
If you plug your DVD player or Amiga to the green RCA input and switch the input to YPbPr, you should get some reaction from the scaler IC.

milika, 2019-03-30:
Well no luck, no picture. And it draws 0.17a
It drew 0.5a before, so in some of those steps something broke.

ramapcsx2, 2019-03-30:
Yep, that'd explain the chip not replying.
You could check if 3.3V get to the scaler chip. Very near the location for the I2C pins on it, there is capacitor C39. One side of it should read 3.3V (the other is Gnd).
If you don't get 3.3V there, check the power trace that leads to the scaler IC area.
If you do get 3.3V, then power is there and the onboard MCU should be able to control the scaler.
In that case, maybe try whatever else you can think of, but it does look like the scaler is fried then :(

Note: I never managed to fry any of my boards, and I did some pretty rough things.. I'd say the scaler IC is pretty well protected.

milika, 2019-03-30:
Thanks,
tried a bunch of stuff, VCC GND, fine on all the chips, TV5725 seems no shorts, no leakage, but its not drawing any power. Even thought it has VCC. Memory chip has VCC and GND.
I found out that one of my D1 mins, now wont work on 5v, only on 3.3, so it must have fried something when that happened. But that one was connected directly to PSU, not even to the board...

[7 more replies]

closed 2019-03-30 by milika