ramapcsx2/gbs-control #63: MDA v-sync not workin

BenjaminRenz, 2019-03-29:
Good evening,
I'm trying to use a gbs8200 to create an adapter from mda/hercules to vga. This mode is not supported by the official adapter because h-frequency is 18.4 kHz and therefore out of range according to the manual. But with this firmware I kind of got it working, but the picture is scrolling vertically through the screen from top to bottom, so vsync is not working. The signal is there electrically, I checked with an oscilloscope so it's probably not caused by that.

I used an esp-01 as an adapter (the black one with 1Mbyte flash). I can access the website and change settings, but can't access the serial monitor over uart because all pins of the board are used for i2c, debug and led. (It only has 4 broken out gpio pins, but I might be able to connect some more but that would require some smd soldering)

Get Video Timings returns:
htotal: 1802
HS ST/SP       :     8 108
HB ST/SP(dis):    1696 352
HB ST/SP       :    1696 256
-------
vtotal 1000
VS ST/SP      :     1    5
VB ST/SP(dis):    990 28
VB ST/SP      :     4 10
IF_VB_ST/SP:      18 20

The stange thing is that it doesn't matter if I have the v-sync signal connected or not.
All options are zero except:
frame time lock, auto gain, scanlines,line_filter, peaking
are all =1

Is mda/hercules even possible with this chip and does someone have a recommendation on what to try to get sync workin?
Greetings
Benjamin

ramapcsx2, 2019-03-30:
Hey,
the timings shouldn't be a problem, if you connect your source using separate H/V Sync.
I need to see some logs of what gbscontrol detects.
You can use the web ui log window to grab some details, but since this is using websockets, it doesn't always capture everything. You may want to retry a few times and see what you get.

- Please disable frame time lock. This is meant for CSync mode, and is there to defeat the single buffer VSync tear line, so it doesn't apply here.

- The get video timings command displays output information for the loaded preset, not input timings (I should rename this, I think).

- Even if you can't capture good logs, the main thing to watch out for is whether it runs "RGBHV bypass" mode, or not. You can tell it's in bypass mode when none of the output resolution presets seem to change anything.

BenjaminRenz, 2019-03-30:
> Hey,
> the timings shouldn't be a problem, if you connect your source using separate H/V Sync.
> I need to see some logs of what gbscontrol detects.
> You can use the web ui log window to grab some [...]

Yes it says RGBHV bypass,
maybe that would explain what happened on the first monitor which is an older model.
Just a hour ago this LCD died, but another newer monitor complains about input format out of spec, so the gbs seems to pass the frames through without modifying the timing. I would have guessed that it has SRAM to store some frames, but is it even able to output at different framerates?

Frame thime lock disabled does not seem to change anything.

BenjaminRenz, 2019-03-30:
If I interpret the manual corectely its impossible to change the timings in bypass mode right?
 ```
5.8.3 Timing generation for bypass mode
For bypass mode, we must keep the input video stream’s line rate and frame rate, but we can adjust
the horizontal and vertical sync position and blanking width and position:
```

So I would need to get it from Mda/Hercules
v: 50Hz h: 18.4kHz  r: 720 × 348 to something like 
v: 85Hz h:  53.7KHz r: 800 x 600
or any other modern vga timing which has any chance to be accepted by the monitor.

BenjaminRenz, 2019-03-30:
I have solderd additional pins to the esp and now have a serial interface working. When it's rebooted it outputs:
```
starting
(WiFi): still connecting..
userprefs open ok
preset preference = 2
frame time lock = 0
preset slot = 1
frame lock method = 0
auto gain = 1
scanlines = 1
component output = 0
deinterlacer mode = 1
line filter = 1
peaking = 1
pal force 60 = 0
i2c: 1A
       <reset>
Activity detected, input: RGBS/HV
VSync: present
HSync: present
RGBHV bypass
ADC gain: keep previous
7B 7B 7B
post preset done (preset id: 1)
(WiFi) AP mode (SSID: gbscontrol, pass 'qqqqqqqq'): Access 'gbscontrol.local' in your browser

```

ramapcsx2, 2019-04-01:
Okay, I haven't documented this properly yet, but there are basically 2 operating modes for gbscontrol:
- RGBHV bypass
- Upscaler mode

Users can't select between the modes yet. Instead, bypass mode is selected whenever separate Sync is provided, and upscaler mode is selected when a form of CSync is provided.
So all you need to do is modify your cable, in case your source provides CSync.

I have some preliminary code in place to do upscaling mode with separate Sync, but it needs more work yet.
If you want to try it, change "uopt->preferScalingRgbhv" to "1": https://github.com/ramapcsx2/gbs-control/blob/master/gbs-control.ino#L3854

ramapcsx2, 2019-04-01:
The high horizontal rate might be a problem for the monitor still. If that happens, I'll show you how to disable the field rate adjustment feature.

BenjaminRenz, 2019-04-01:
> Okay, I haven't documented this properly yet, but there are basically 2 operating modes for gbscontrol:
>
>     * RGBHV bypass
>
>     * Upscaler mode
>
>
> Users can't select between the modes yet. Instead, [...]

I'll test the the "software" solution first, because what I've found as Vsync/Hsync to Csync requieres components I don't have at hand right now http://www.epanorama.net/circuits/vga2rgbs.html , maybe a moderm microconroller should be able to do that too, but it would be a bit wastefull.

I'll try that tommorow and report back.

BenjaminRenz, 2019-04-02:
I'm not 100% sure if I flashed the esp corectely, but after changing the uopt->preferScalingRgbhv to = 1 it still showes RGBHV bypass, but I will test that again tomorrow. Unfortunately I don't have that much time at the moment, but that should get better on the weekend. Can you tell from a dump (d) if the mode was changed as expected?
My monitor still shows a black picture, but I will try to find an older model because they seem to be more forgiving when it comes to missmatched timing and still show a wandering picture.

Would it be possible to change those settings on the fly by manipulation the registers directely? If I've seen that corectely you have some commands to read/write to the directely?

https://github.com/ramapcsx2/gbs-control/blob/master/gbs-control.ino#L4720

The only useful document from trueview seems to be their internal programming guide.

ramapcsx2, 2019-04-03:
It's possible I broke the initial implementation. I'll check it when I have some time.

You're right though, the chip has to be programmed via its registers.
To get an idea of the process, this function enables the bypass mode. 
https://github.com/ramapcsx2/gbs-control/blob/master/gbs-control.ino#L3156
This mode requires the least programming, as there's no VDS processor, no deinterlacing pipe and no Input Formatter.

BenjaminRenz, 2019-04-03:
> 
> 
> It's possible I broke the initial implementation. I'll check it when I have some time.
> 
> You're right though, the chip has to be programmed via its registers.
> To get an idea of the process, this function enables the bypass mode.
> https://github.com/ramapcsx2/gbs-control/blob/master/gbs-control.ino#L3156
> This mode requires the least programming, as there's no VDS processor, no deinterlacing pipe and no Input Formatter.
> You can reference the GBS::xxx functions with the register definitions.
> 
> The "s" and "t" commands you found allow setting an entire register, or toggling a single bit.
> Format for setting:
> s02s1bsff (set segment 02, register 0x1b, value 0xff)
> Format for bit toggle:
> t02t1bt7 (toggle segment 02, register 0x1b, bit 7)
> Leading zeros get ignored to some reasonable extent. My parser isn't the best ;p

I can now confirm that the esp was reprogrammed and still shows bypass mode. 
Thanks for the explanation on the registers, I'll try to start from the bypass mode and read the programming guide a bit, thanks 👍

ramapcsx2, 2019-04-04:
I checked the code and it surely was too limiting for your source format. Try the latest master for a much better chance of the mode actually working.
I'm not sure what details your PAL mode will expose though, still working on it ;p

BenjaminRenz, 2019-04-04:
> 
> 
> I checked the code and it surely was too limiting for your source format. Try the latest master for a much better chance of the mode actually working.
> I'm not sure what details your PAL mode will expose though, still working on it ;p

Testing results:

```
starting
(WiFi): still connecting..
userprefs open ok
preset preference = 0
frame time lock = 0
preset slot = 1
frame lock method = 0
auto gain = 0
scanlines = 0
component output = 0
deinterlacer mode = 0
line filter = 1
peaking = 1
pal force 60 = 0
i2c: 1A
<reset>
Activity detected, input: RGBS/HV
VSync: present
HSync: present
RGBHV bypass 
post preset done (preset id: 1)
60Hz EDTV 
post preset done (preset id: 1)
large diff
ABHT: retry
large diff
ABHT: retry
large diff
ABHT: retry
large diff
ABHT: retry
large diff
ABHT: retry
large diff
ABHT: retry
large diff
ABHT: retry
large diff
ABHT: give up
```

I can now see a picture on the monitor and it no longer complains about input out of range, but the picture freezes and it's not aligned (picture underway)...

ramapcsx2, 2019-04-04:
Awesome, try sending "t5t11t5" (and update your Datenkopf xD).

ramapcsx2, 2019-04-04:
New update adds some distinction between 60Hz and 50Hz. Your mode should load the PAL preset now, which should fit the requirement a little better, I hope.
I also changed the H-PLL charge pump current, in case the presets have set it too high for these broad applications.

BenjaminRenz, 2019-04-04:
With t5t11t5 and plugged in Datenkopf...

[image]

BenjaminRenz, 2019-04-04:
> New update adds some distinction between 60Hz and 50Hz. Your mode should load the PAL preset now, which should fit the requirement a little better, I hope.
> I also changed the H-PLL charge pump current, in case the presets have set it too high for these broad applications.
Test with latest firmware:

```
starting
(WiFi): still connecting..
userprefs open ok
preset preference = 0
frame time lock = 0
preset slot = 1
frame lock method = 0
auto gain = 0
scanlines = 0
component output = 0
deinterlacer mode = 0
line filter = 1
peaking = 1
pal force 60 = 0
i2c: 1A
<reset>
Activity detected, input: RGBS/HV
VSync: present
HSync: present
RGBHV bypass 
post preset done (preset id: 1)
60Hz EDTV 
post preset done (preset id: 1)
large diff
ABHT: retry
large diff
ABHT: retry
large diff
ABHT: retry
large diff
ABHT: retry
large diff
ABHT: retry
large diff
ABHT: retry
large diff
ABHT: retry
large diff
ABHT: give up
(WiFi) AP mode (SSID: gbscontrol, pass 'qqqqqqqq'): Access 'gbscontrol.local' in your browser
```

Now there seems to be artifacting in the lower half of the picture:
[image]
Closeup on artifacts (some colorful dots):
[image]

ramapcsx2, 2019-04-04:
Okay, it's not picking up the 50Hz mode. Can you share a few lines of the info mode log ("i")?

BenjaminRenz, 2019-04-04:
The lower half of the picture does change/update if the input mda cable is reconnected.
```
h: 511 v: 193 PLL.00 A:7b7b7b S:0b.10 I:00 D:0666 m:14 ht:2557 vt:369 hpw: 331 s:ff W:31 F:36192 L:4
h: 511 v: 193 PLL.00 A:7b7b7b S:0b.10 I:00 D:0666 m:14 ht:2558 vt:369 hpw: 332 s:ff W:31 F:36192 L:4
h: 511 v: 193 PLL.00 A:7b7b7b S:0b.10 I:00 D:0664 m:14 ht:2558 vt:369 hpw: 331 s:ff W:31 F:36192 L:5
h: 511 v: 193 PLL.00 A:7b7b7b S:0b.10 I:00 D:0666 m:14 ht:2559 vt:369 hpw: 331 s:ff W:31 F:36192 L:4
h: 511 v: 193 PLL.00 A:7b7b7b S:0b.10 I:00 D:0666 m:14 ht:2558 vt:369 hpw: 332 s:ff W:31 F:36192 L:4
h: 511 v: 193 PLL.00 A:7b7b7b S:0b.10 I:00 D:0666 m:14 ht:2557 vt:369 hpw: 332 s:ff W:31 F:36192 L:5
h: 511 v: 193 PLL.00 A:7b7b7b S:0b.10 I:00 D:0666 m:14 ht:2558 vt:369 hpw: 331 s:ff W:31 F:36192 L:4
h: 511 v: 193 PLL.00 A:7b7b7b S:0b.10 I:00 D:0666 m:14 ht:2558 vt:369 hpw: 332 s:ff W:31 F:36192 L:4
h: 511 v: 193 PLL.00 A:7b7b7b S:0b.10 I:00 D:06e7 m:14 ht:2557 vt:369 hpw: 331 s:ff W:31 F:36192 L:5
h: 511 v: 193 PLL.00 A:7b7b7b S:0b.10 I:00 D:06e7 m:14 ht:2558 vt:369 hpw: 331 s:ff W:31 F:36192 L:5
h: 511 v: 193 PLL.00 A:7b7b7b S:0b.10 I:00 D:06e7 m:14 ht:2557 vt:369 hpw: 331 s:ff W:31 F:36192 L:5
h: 511 v: 193 PLL.00 A:7b7b7b S:0b.10 I:00 D:0666 m:14 ht:2558 vt:369 hpw: 331 s:ff W:31 F:36192 L:5
h: 511 v: 193 PLL.00 A:7b7b7b S:0b.10 I:00 D:0666 m:14 ht:2558 vt:369 hpw: 332 s:ff W:31 F:36192 L:4
h: 511 v: 193 PLL.00 A:7b7b7b S:0b.10 I:00 D:0666 m:14 ht:2558 vt:369 hpw: 332 s:ff W:31 F:36192 L:4
h: 511 v: 193 PLL.00 A:7b7b7b S:0b.10 I:00 D:0666 m:14 ht:2557 vt:369 hpw: 331 s:ff W:31 F:36192 L:5
h: 511 v: 193 PLL.00 A:7b7b7b S:0b.10 I:00 D:0666 m:14 ht:2559 vt:369 hpw: 332 s:ff W:31 F:36192 L:4
h: 511 v: 193 PLL.00 A:7b7b7b S:0b.10 I:00 D:06e7 m:14 ht:2558 vt:369 hpw: 332 s:ff W:31 F:36192 L:5
h: 511 v: 193 PLL.00 A:7b7b7b S:0b.10 I:00 D:0666 m:14 ht:2558 vt:369 hpw: 332 s:ff W:31 F:36192 L:5
h: 511 v: 193 PLL.00 A:7b7b7b S:0b.10 I:00 D:0666 m:14 ht:2557 vt:369 hpw: 332 s:ff W:31 F:36192 L:4
h: 511 v: 193 PLL.00 A:7b7b7b S:0b.10 I:00 D:0666 m:14 ht:2558 vt:369 hpw: 332 s:ff W:31 F:36192 L:4
h: 511 v: 193 PLL.00 A:7b7b7b S:0b.10 I:00 D:0666 m:14 ht:2558 vt:369 hpw: 332 s:ff W:31 F:36192 L:5
h: 511 v: 193 PLL.00 A:7b7b7b S:0b.10 I:00 D:0666 m:14 ht:2558 vt:369 hpw: 332 s:ff W:31 F:36192 L:4
h: 511 v: 193 PLL.00 A:7b7b7b S:0b.10 I:00 D:0666 m:14 ht:2558 vt:369 hpw: 332 s:ff W:31 F:36192 L:5
h: 511 v: 193 PLL.00 A:7b7b7b S:0b.10 I:00 D:0666 m:14 ht:2558 vt:369 hpw: 331 s:ff W:31 F:36192 L:5
h: 511 v: 193 PLL.00 A:7b7b7b S:0b.10 I:00 D:0666 m:14 ht:2558 vt:369 hpw: 331 s:ff W:31 F:36192 L:4
h: 511 v: 193 PLL.00 A:7b7b7b S:0b.10 I:00 D:0676 m:14 ht:2558 vt:369 hpw: 332 s:ff W:31 F:36192 L:4
h: 511 v: 193 PLL.00 A:7b7b7b S:0b.10 I:00 D:0666 m:14 ht:2558 vt:369 hpw: 332 s:ff W:31 F:36192 L:5
h: 511 v: 193 PLL.00 A:7b7b7b S:0b.10 I:00 D:0664 m:14 ht:2559 vt:369 hpw: 332 s:ff W:31 F:36192 L:4
h: 511 v: 193 PLL.01 A:7b7b7b S:0b.10 I:00 D:0666 m:14 ht:2559 vt:369 hpw: 331 s:ff W:31 F:36192 L:4
h: 511 v: 193 PLL.01 A:7b7b7b S:0b.10 I:00 D:0666 m:14 ht:2557 vt:369 hpw: 332 s:ff W:31 F:36192 L:6
h: 511 v: 193 PLL.01 A:7b7b7b S:0b.10 I:00 D:06e7 m:14 ht:2558 vt:369 hpw: 332 s:ff W:31 F:36192 L:4
h: 511 v: 193 PLL.01 A:7b7b7b S:0b.10 I:00 D:0666 m:14 ht:2558 vt:369 hpw: 332 s:ff W:31 F:36192 L:4
h: 511 v: 193 PLL.01 A:7b7b7b S:0b.10 I:00 D:0666 m:14 ht:2559 vt:369 hpw: 332 s:ff W:31 F:36192 L:5
h: 511 v: 193 PLL.01 A:7b7b7b S:0b.10 I:00 D:0666 m:14 ht:2558 vt:369 hpw: 332 s:ff W:31 F:36192 L:4
h: 511 v: 193 PLL.01 A:7b7b7b S:0b.10 I:00 D:0666 m:14 ht:2558 vt:369 hpw: 331 s:ff W:31 F:36192 L:4
h: 511 v: 193 PLL.01 A:7b7b7b S:0b.10 I:00 D:0666 m:14 ht:2559 vt:369 hpw: 331 s:ff W:31 F:36192 L:5
h: 511 v: 193 PLL.01 A:7b7b7b S:0b.10 I:00 D:06c5 m:14 ht:2559 vt:369 hpw: 332 s:ff W:31 F:36192 L:4
h: 511 v: 193 PLL.01 A:7b7b7b S:0b.10 I:00 D:0666 m:14 ht:2558 vt:369 hpw: 332 s:ff W:31 F:36192 L:4
h: 511 v: 193 PLL.01 A:7b7b7b S:0b.10 I:00 D:0666 m:14 ht:2558 vt:369 hpw: 332 s:ff W:31 F:36192 L:5
h: 511 v: 193 PLL.01 A:7b7b7b S:0b.10 I:00 D:0666 m:14 ht:2558 vt:369 hpw: 332 s:ff W:31 F:36192 L:4
h: 511 v: 193 PLL.01 A:7b7b7b S:0b.10 I:00 D:0666 m:14 ht:2558 vt:369 hpw: 332 s:ff W:31 F:36192 L:4
h: 511 v: 193 PLL.01 A:7b7b7b S:0b.10 I:00 D:0666 m:14 ht:2558 vt:369 hpw: 332 s:ff W:31 F:36192 L:5
h: 511 v: 193 PLL.01 A:7b7b7b S:0b.10 I:00 D:0666 m:14 ht:2558 vt:369 hpw: 332 s:ff W:31 F:36192 L:5
h: 511 v: 193 PLL.01 A:7b7b7b S:0b.10 I:00 D:0666 m:14 ht:2557 vt:369 hpw: 331 s:ff W:31 F:36192 L:4
h: 511 v: 193 PLL.00 A:7b7b7b S:0b.10 I:00 D:0666 m:14 ht:2558 vt:369 hpw: 331 s:ff W:31 F:36192 L:6
h: 511 v: 193 PLL.00 A:7b7b7b S:0b.10 I:00 D:0664 m:14 ht:2558 vt:369 hpw: 332 s:ff W:31 F:36192 L:4
h: 511 v: 193 PLL.00 A:7b7b7b S:0b.10 I:00 D:0666 m:14 ht:2558 vt:369 hpw: 331 s:ff W:31 F:36192 L:4
h: 511 v: 193 PLL.00 A:7b7b7b S:0b.10 I:00 D:0666 m:14 ht:2557 vt:369 hpw: 331 s:ff W:31 F:36192 L:5
h: 511 v: 193 PLL.00 A:7b7b7b S:0b.10 I:00 D:0666 m:14 ht:2558 vt:369 hpw: 332 s:ff W:31 F:36192 L:4
h: 511 v: 193 PLL.00 A:7b7b7b S:0b.10 I:00 D:0666 m:14 ht:2558 vt:369 hpw: 331 s:ff W:31 F:36192 L:5
h: 511 v: 193 PLL.00 A:7b7b7b S:0b.10 I:00 D:0666 m:14 ht:2559 vt:369 hpw: 331 s:ff W:31 F:36192 L:5
h: 511 v: 193 PLL.00 A:7b7b7b S:0b.10 I:00 D:0666 m:14 ht:2558 vt:369 hpw: 331 s:ff W:31 F:36192 L:4
h: 511 v: 193 PLL.00 A:7b7b7b S:0b.10 I:00 D:0666 m:14 ht:2558 vt:369 hpw: 332 s:ff W:31 F:36192 L:4
[... cut ...]
h: 511 v: 193 PLL.00 A:7b7b7b S:0b.10 I:00 D:0666 m:14 ht:2558 vt:369 hpw: 332 s:ff W:31 F:36192 L:5
h: 511 v: 193 PLL.00 A:7b7b7b S:0b.10 I:00 D:0666 m:14 ht:2559 vt:369 hpw: 332 s:ff W:31 F:36192 L:4
h: 511 v: 193 PLL.00 A:7b7b7b S:0b.10 I:00 D:0666 m:14 ht:2558 vt:369 hpw: 332 s:ff W:31 F:36192 L:4
h: 511 v: 193 PLL.00 A:7b7b7b S:0b.10 I:00 D:06e7 m:14 ht:2558 vt:369 hpw: 331 s:ff W:31 F:36192 L:5
h: 511 v: 193 PLL.00 A:7b7b7b S:0b.10 I:00 D:0666 m:14 ht:2558 vt:369 hpw: 331 s:ff W:31 F:36192 L:5

```

ramapcsx2, 2019-04-04:
Update incoming :)

BenjaminRenz, 2019-04-04:
Also works on the newer monitor now, the upper half of the picture is updating while the lower half stays frozen.

ramapcsx2, 2019-04-04:
Update pushed.
It will fix the issue you see and hopefully do the other automatic adjustments in spec with your monitors.

From here on, as long as you get a mostly okay image, you can do all sorts of image manipulation via the web interface (as scaling mode now works).

BenjaminRenz, 2019-04-04:
> 
> 
> Update pushed.
> It will fix the issue you see and hopefully do the other automatic adjustments in spec with your monitors.
> 
> From here on, as long as you get a mostly okay image, you can do all sorts of image manipulation via the web interface (as scaling mode now works).

Nice, seems to work now.

ramapcsx2, 2019-04-04:
Okay, great!
I assume you'll have to adjust scaling and then image position quite a bit, to get everything on screen.
The preset loaded is optimized for real PAL, then stretched out a bit for game consoles that usually don't make use of all available pixels / lines.

In effect this just means that sharpness won't be quite optimal, but your use case sounds more like you just wanted to be able to read what's being displayed, not enjoy some Mario ;)

BenjaminRenz, 2019-04-04:
> Okay, great!
> I assume you'll have to adjust scaling and then image position quite a bit, to get everything on screen.
> The preset loaded is optimized for real PAL, then stretched out a bit for game [...]

Exactly, it looks great in comparison to the old crt :),
the only thing that is still noticable is a "bar" which is about 5% of the screensize and scrolling from top to bottom where it shifts all pixels just a little bit. It takes about half a secont from top to bottom and is horizontal. But this could be also present on the source signal or due to the lack of modifications I've done to the gbs board.
This is not a problem for me, just if it would be an easy fix it might be good if someone wants to play games on MDA. Could also upload a video for the ghosting bar.

[image]

Setting the "d" output for scaling options to fit on the full hd panel:

```
0x2C, // s0_40
0x95, // s0_41
0x0, // s0_42
0x19, // s0_43
0x25, // s0_44
0x11, // s0_45
0x7F, // s0_46
0x17, // s0_47
0xEB, // s0_48
0xB, // s0_49
0x4, // s0_4A
0x0, // s0_4B
0x0, // s0_4C
0x2A, // s0_4D
0x0, // s0_4E
0x3C, // s0_4F
0x0, // s0_50
0x0, // s0_51
0x67, // s0_52
0x0, // s0_53
0x0, // s0_54
0x0, // s0_55
0x0, // s0_56
0x0, // s0_57
0x0, // s0_58
0xFF, // s0_59
0x0, // s0_5A
0x0, // s0_5B
0x0, // s0_5C
0x0, // s0_5D
0x0, // s0_5E
0x0, // s0_5F
0x24, // s0_90
0x0, // s0_91
0x47, // s0_92
0x54, // s0_93
0x0, // s0_94
0x3E, // s0_95
0x32, // s0_96
0x0, // s0_97
0x0, // s0_98
0x0, // s0_99
0x0, // s0_9A
0x0, // s0_9B
0x0, // s0_9C
0x0, // s0_9D
0x0, // s0_9E
0x0, // s0_9F
0x20, // s1_0
0xE0, // s1_1
0x64, // s1_2
0x0, // s1_3
0x0, // s1_4
0x0, // s1_5
0x0, // s1_6
0x0, // s1_7
0x0, // s1_8
[... 352 lines cut ...]
0x0, // s5_59
0x1, // s5_5A
0x0, // s5_5B
0x3, // s5_5C
0x2, // s5_5D
0x0, // s5_5E
0x0, // s5_5F
0x0, // s5_60
0x0, // s5_61
0x0, // s5_62
0xF, // s5_63
0x0, // s5_64
0x0, // s5_65
0x0, // s5_66
0x0, // s5_67
0x0, // s5_68
0x0, // s5_69
0x0, // s5_6A
0x0, // s5_6B
0x0, // s5_6C
0x0, // s5_6D
0x0, // s5_6E
0x0, // s5_6F
};
```

BenjaminRenz, 2019-04-04:
I got a slow motion capture of it with my phone where 2 seconds captured are about 8 seconds video
[slowmotion.zip](https://github.com/ramapcsx2/gbs-control/files/3045118/slowmotion.zip)

ramapcsx2, 2019-04-04:
Hmm, this looks unfamiliar. 
It might be an artifact of the rather poor H-PLL lock (I noticed that the display is slightly wobbly, a sign of the Sync Processor not being in best phase).
To check the SP phase, you could send "vvvv" for a few times, check the readout to see how this works (it loops).
There will be a spot where the display wobble gets worse, and hopefully a spot where it gets better.
Automating this process is a long term goal of this project.

Also, the source unfortunately has that mesh effect that shows ADC / horizontal scaling imperfections well.
If this annoys you (like it does me :p), you can try in/decreasing horizontal scaling a few notches, until it reduces or disappears completely.
Automating that process is an even longer term goal ;p

BenjaminRenz, 2019-04-05:
> Hmm, this looks unfamiliar.
> It might be an artifact of the rather poor H-PLL lock (I noticed that the display is slightly wobbly, a sign of the Sync Processor not being in best phase).
> To check the [...]

changing the H-PLL lock does not seem to help with the scrolling through distortion. It makes the rest of the picutre more stable though. 
Inreasing the PLL divider to A1F does nearly completely eliminated the vertical bars in the picture.

Final Result (just experimented if the potentiometers change something 😁, the dark lines are not as noticable on the monitor, my camera seems to enhance the contrast there quiet a bit):
[image]

Tanks for the great help, awsome project 👍

closed 2019-04-05 by BenjaminRenz

ramapcsx2, 2019-04-05:
Alright, glad you like it.

If you just make sure not to cheat on that diesel test, that'll be great ;)

userx14, 2022-01-21:
> Impressive - I'm repairing an old Bosch tester at the moment and have similar problems. My old screen is not repairable anymore and I replaced it with a TFT. The output is a composite signal and the destination a 800:600 TFT screen. It's possible, that Gutmann used the same technology to connect their screens. If I use a 4:3 resolution I have also problems with the v-sync.

800x600 sounds like it is to high of a resolution to be mda though, more like ega or vga.

melo1, 2022-01-21:
The output is something like 640x40 at about 60Hz.
It's an composite signal which I send over the green channel (sync on green).
With original firmware I get a picture but it's not stable.
With gbs control it's much better but the 4:3 resolutions are rolling from top to bottom.
At the moment I have no idea how to solve the problem and have not tested very much.
The tester is very huge an I have to go there with my notebook to play around.

userx14, 2022-01-23:
> The output is something like 640x40 at about 60Hz. It's an composite signal which I send over the green channel (sync on green). With original firmware I get a picture but it's not stable. With gbs [...]

Oh sorry, I missunderstood, you were refering to the monitor you want to connect the signal from the gbs to.

> The tester is very huge an I have to go there with my notebook to play around.

I think it is possible to connect to the esp over wifi and access the website with a phone.
And if I read the code correctelly one can use "x" to toggle uopt->preferScalingRgbhv.
https://github.com/ramapcsx2/gbs-control/blob/b9265e4025730aff7e0a4dbb82ac862eb9ba1b6a/gbs-control.ino#L9087-L9097
But I have not tested it with the lastest version yet.

melo1, 2022-01-23:
> Oh sorry, I missunderstood, you were refering to the monitor you want to connect the signal from the gbs to.
>
>
> I think it is possible to connect to the esp over wifi and access the website with a [...]

Thank you - I'll check that.

andreacampanella, 2022-01-28:
Hello, 
I've been pointed to this ticket from another one. 

It seems you all be having success getting MDA into the GBS, I'm just wondering how am I supposed to connect it, does R,G,B,H,V goes straight into the VGA input? 

isn't MDA 5v and VGA 0.7v max?

userx14, 2022-01-29:
@andreacampanella
I'm using this adapter, please don't look to close at the soldering quality 😄 .
I've connected all 3 color inputs (r,g,b) together and used a 330 ohm resistor to connect with the video pin of the hercules card.
The build is about 3 years ago, so I don't remember every detail, but as far as I know the though process was as follows:

The rgb input should be terminated with 75 ohm, so when uising the 330 ohm resistor one would get a voltage of 0.92 V from the 5V mda voltage. In the datasheed it says that the default adc range is 0 mV - 770 mV, but this full scale range seems to be adjustable up to 1 V. Since I need a black and white image 920 mV worked for me.

As far as I know Hsync and Vsync are TTL for both VGA and MDA. Even though I was too lazy to modify the 500 ohm termination with the additional 100 ohm resisor (https://github.com/ramapcsx2/gbs-control/wiki/Inputs:-SCART-adapter,-optional-sync-stripper) it still seems to work, but I guess that can vary from board to board.

H sync and V sync are just directelly connected (grey, yellow).
[image]
[image]

Best
Benjamin

A-Small-Mice, 2022-09-02:
@andreacampanella

To get MDA, I used the tool named GBS-CGA/EGA on the following links:
https://github.com/jtsiomb/gbs-cgaega
https://github.com/A-Small-Mice/gbs-cgaega-pld-files-for-improved-colors

GBS Control tries to synchronize at 18 khz, but it should fail to get there in the end.
I figured out a good solution that as follows.

1. When the GBS Control first begins to get the MDA signal, the syncronization is normal for about 2 seconds.
2. Turn off 'Sync Watcher' at that time in the control panel of GBS Control.

* Changing resolution temporarily restores  the syncronization.
* This must be done before GBS Control completely loses synchronization.

A-Small-Mice, 2022-09-02:
@andreacampanella

To get MDA/Hercules, I used a tool named GBS-CGA/EGA on the following links:
https://github.com/jtsiomb/gbs-cgaega
https://github.com/A-Small-Mice/gbs-cgaega-pld-files-for-improved-colors

GBS Control tries to synchronize at 18 khz, but it should fail to get there in the end.
I figured out a good solution that as follows.

When the GBS Control first begins to get the MDA signal, the syncronization is normal for about 2 seconds.
Turn off 'Sync Watcher' at that time in the control panel of GBS Control.
Changing resolution temporarily restores the syncronization.
This must be done before GBS Control completely loses synchronization.