ramapcsx2/gbs-control #66: Physical Config Slot Switch

ILAHWWINC, 2019-04-20:
Will it be possible to implent a switch to swap between two config sets.

I use two different config slots anderen it would be nice to change without using another device an the gui over wlan.

ramapcsx2, 2019-04-22:
You mean a physical switch?
We're limited by the number of easily available hardware switches:
- usually a reset and a "flash" button on the ESP8266 board
- 4 buttons on the GBS that aren't wired to anything by default

The 4 buttons on the GBS can be hooked up to the ESP8266 and there is code there to support them.
If you find 4 available / unused inputs on the ESP8266 board, the code behind the #HAVE_BUTTONS define can be used.
https://github.com/ramapcsx2/gbs-control/blob/master/gbs-control.ino#L88
Just need to adapt the pin mapping and test, since this was originally meant for Arduino (it's quite old and unused).

ILAHWWINC, 2019-04-22:
Thank you for answering. Yes, a physical switch. 

It could be connected to the ESP Board only and be fixed to a casing for example.

I want to use two settings, 1. Amiga Workbench Hires Laced, 2. Amiga Games Default. They need different picture size and position. It would be nice to switch between them old fashioned was. 
1. Click 2. Clack

ramapcsx2, 2019-04-22:
Simple enough, but you should add this yourself then :)
You'll want to hook up your switch to a free GPIO pin on your ESP dev board.
Check Google for Arduino physical switch tutorials if you need help there.
Then in the gbscontrol sketch, you can add a switch status check in the function called "syncwatcher" (it runs every few milliseconds), so the switch position gets constantly updated.
You probably need to override the automatic preset assignment. 
I guess if you do this somewhere in setup(), it should work: `uopt->presetPreference = 2;`

If your switch reads "low", do` uopt->presetSlot = 1;`
If it reads "high", do `uopt->presetSlot = 2;`
(Or similar, depends where your have your preset stored.)

ILAHWWINC, 2019-04-24:
I will give it a try. Please check wether i am right with my explanations.

Hardware:  5V - #Switch# - GND (over1kOhm) _and_ GPIO Pin 2 (Pin D4 on NodeMCU)

Software: 1. to be copied in the area syncwatcher

`const int switchPin = 2;
int switchState = 0;
 
void setup() {
pinMode(switchPin, INPUT);
}
 
void loop(){
switchState = digitalRead(switchPin);
if (switchState == HIGH) {
digitalWrite uopt->presetSlot = 2;
}
else {
digitalWrite uopt->presetSlot = 1;
}
}`

Software: 2. to be copied in the area setup ()

`uopt->presetPreference = 2;`

Should that work?

ramapcsx2, 2019-04-24:
Use 3.3V instead of 5V for your HIGH state. Most everything on the GBS and the ESP8266 is 3.3V anyway.

It's better to put the `switchState = digitalRead(switchPin);` logic within the runSyncWatcher() function, so it doesn't run each loop.

Replace `digitalWrite uopt->presetSlot = 2;` and `digitalWrite uopt->presetSlot = 1;` with just `uopt->presetSlot = 2;` and `uopt->presetSlot = 1;`. These just set a variable that tells the applyPresets() function which slot the user prefers.

Once you have all that, you probably want to switch presets as soon as you flick the switch, right?
If so, you'll have to run applyPresets(detectedVideoMode) right after your new code in runSyncWatcher().
(The detectedVideoMode variable is a local to runSyncWatcher(), so it only exists there.)

It should work! :)

ILAHWWINC, 2019-04-25:
This is very difficult for me because i am not common with programming.

Is it correct to copy the code fragment above in the gbs-control.ino after the line
(3526) uint32_t runSyncWatcher() ?

What should i do with the applyPresets(detectedVideoMode) command. 

Can you please give me some more hints.

Thanks

ramapcsx2, 2019-04-26:
I'll code it up for you but it'll take a bit. Pretty busy :p

ILAHWWINC, 2019-04-26:
Great news, thanks. No hurry.

ramapcsx2, 2019-04-26:
```
static const int switchPin = 13; // Pin D7 on NodeMCU (Pin D4 / GPIO 2 isn't suitable)
  // pinMode() should better be in setup(), but here works for demonstration
  pinMode(switchPin, INPUT_PULLUP); // or INPUT, if you have an external pullup/pulldown resistor
  static uint8_t oldSwitchState = 1;
  int switchState = 1;
  if (detectedVideoMode > 0) {
    switchState = digitalRead(switchPin);
    if (switchState != oldSwitchState) {
      // oldSwitchState is still what it was the last time this function ran
      // so when switchState != oldSwitchState, the switch must have changed!
      uopt->presetPreference = 2;
      if (switchState == 1) {
        SerialM.println("slot 2");
        uopt->presetSlot = 2;
      }
      else if (switchState == 0) {
        SerialM.println("slot 1");
        uopt->presetSlot = 1;
      }
      saveUserPrefs();
      applyPresets(detectedVideoMode);
    }

    oldSwitchState = switchState; // now both are the same. 
  }
```

Put this code here: https://github.com/ramapcsx2/gbs-control/blob/master/gbs-control.ino#L4024
so it runs last in runSyncWatcher().
I recommend changing the GPIO to 13 (D7), because GPIO 2 has a special function and isn't good for use in a switch.

If I haven't missed anything, it should work then :)

ILAHWWINC, 2019-04-27:
Works like a charm. Many thanks for the fast and personal support.

ramapcsx2, 2019-04-27:
Sure, enjoy :p

closed 2019-04-27 by ramapcsx2

ILAHWWINC, 2022-06-11:
Will it be possible to inject this code into the last .ino file?
I would like to keep using this switch with the latest build.
If it is possible can you tell me where to put it?