ramapcsx2/gbs-control #67: Custom preset not loading as defauly on reset

icrutt, 2019-04-26:
I've flashed the gbs-control code onto an Adafruit HUZZAH ESP8266 board and connected it to my v.4 GBS-8200 board (black buttons). 

I'm using it to convert an RGBs signal from an Acorn BBC Micro, output to VGA for display on a 60Hz Dell LCD monitor.

I can adjust the gbs-control configuration using the web interface over WiFi to get a good picture. However, even if I save the configuration to custom preset slot, and leave that slot selected, the system defaults to the built-in 1280x960 preset on reset. I can then select the custom preset and it loads fine. 

Am I correct that the selected custom slot should load by default on reset? This is the output from the board when resetting:

```
starting
(WiFi): IP: 192.168.1.110 Hostname: gbscontrol.local
userprefs open ok
preset preference = 2
frame time lock = 0
preset slot = 1
frame lock method = 0
auto gain = 0
scanlines = 0
component output = 0
deinterlacer mode = 0
line filter = 1
peaking = 1
pal force 60 = 1
i2c: E6
<reset>
Activity detected, input: RGBS/HV

Format change: <stable>
60Hz 
(custom)
userprefs.txt opened
loading from preset slot 1: no preset file for this source
pal forced 60hz: apply vshift
post preset done (preset id: 1)
shift down, vlines: 34
SOG Lvl: 7
userprefs.txt opened
loading from preset slot 1: /preset_pal.1
post preset done (preset id: 1) (custom)
.shift down, vlines: 34
SOG Lvl: 7
```
Any help would be gratefully received. Thanks.

ramapcsx2, 2019-04-26:
Hm, try saving to any slot and then resetting the ESP8266 (not changing anything after you hit save).
I think you may have been changing slots, because the log numbering may not be the same label as the UI.

icrutt, 2019-04-29:
This hasn't made any difference, unfortunately. I saved to slot 2, and immediately reset:
```
saving to preset slot  Restart

Format change: <stable>
60Hz 
(custom)
userprefs.txt opened
loading from preset slot 2: no preset file for this source
pal forced 60hz: apply vshift
post preset done (preset id: 1)
shift down, vlines: 34
SOG Lvl: 7
```

ramapcsx2, 2019-04-30:
It could be a bug with the force 60 Hz option. Could you try disabling that, saving a new preset, then restarting?

icrutt, 2019-04-30:
Thanks - disabling the Force 60Hz option seems to have done the trick.

ramapcsx2, 2019-04-30:
Good to know. It should be fixed now :)

ramapcsx2, 2019-05-04:
Fixed.

closed 2019-05-04 by ramapcsx2