ramapcsx2/gbs-control #68: More options for the scanlines

TomTheDragon, 2019-05-26:
I am really happy with this custom firmware so far and the picture is really much better then the stock FW :)
I was actually kind of surprised to see the scanlines option, since i actually planed to do some kind of hardware mod with a scanline generator. Unfortunately are there no options to change size and/or position of the scanlines, so that they always look kind of "off".

ramapcsx2, 2019-05-29:
Hey,
options for scanlines would be nice, so I'm considering adding some.
It's low priority though and there's not a lot of things to tweak.
Scanline position or size for example can't be changed.

What you're looking for may be achieved by using a different output resolution. That has a big effect on the look of them, all depending on your display as well.

TomTheDragon, 2019-06-12:
I already played around with positions and different resolutions but it always seems off. I however got the best results with higher resolutions.
I think it would be the best, if we could move and change the size of the scanlines independently from the rest of the picture.

ramapcsx2, 2019-06-13:
The effect is a hack. Scanlines appear when "deinterlacing" 2 fields where one field has no data.
Due to this, the scanlines are fixed in position and size, though the appearance can be tweaked with some blending options.

ramapcsx2, 2020-02-13:
I've recently added a scanline strength option. It's a start :)

TomTheDragon, 2020-02-19:
Thanks for the new options! :)
I just updated my ESP with the newer code and it looks quite amazing now ;)
But i think the bigger reason is the inclusion of the new resolution 480p.
Its pretty much perfect as long i dont change picture size in height. It just brakes apart if i do, like it did before.

Here a comparison between my old LCD and a Sony PVM (CRT), running SNES over component
[image]

Close up of the LCD
[image]

Close up of the PVM
[image]

ramapcsx2, 2020-02-19:
That's quite a good result, really like that :)

AlfredoDore, 2025-06-05:
I'm thinking about buying one and I'm researching the subject, I have an LCD monitor with analog inputs, both S-Video, RCA, component and VGA, with that I could use a legitimate analog output, only the screen adjustment and CRT-type scanlines would be left, I really liked the result, couldn't you add this as a default value? I intend to use the Wii component for VGA output and depending on the game, if it's a virtual console or emulator that are running for example on the PS1, use the S-Video output.