ramapcsx2/gbs-control #70: NTSC 720p Scanline glitch.

tomudo, 2019-06-30:
code committed Jun 29, 2019 - a82f2457be99ee7d2f141df29c24ca5a07bc1b06

The screen in NTSC 720p mode distorts every 30 secs when actives scanline mode. The default setting has no issue. 

Additional Note:
@ line 2982, code instruction below these lines won't reach
    else {
      return;
    }

ramapcsx2, 2019-06-30:
I'll check scanlines in a bit.

What is the problem with the code you mention? I don't see it.
The loop is intended to accumulate readings, but exit as soon as one of the error conditions hit.
It does work that way, too.
Am I missing something? :p

tomudo, 2019-06-30:
Thank you for your reply. I tried on PS2 system with component cable. I will look further for what cause it is.
And sorry for that code. I am dumb. :P

closed 2019-06-30 by tomudo

ramapcsx2, 2019-06-30:
Nah, thanks for the feedback anyway! :)

tomudo, 2019-07-01:
I decreased HTotal by hit 'A' in serial console that change HTotal from 2400 to 2398, which eliminate distrotion for NTSC 480i. However, scanline in NTSC 480p is only brighten , scanline effect not applied, and have a vertical green strap in right side of screen.

ramapcsx2, 2019-07-01:
This doesn't sound right at all. I can go wild with the timing parameters in 720p output and there aren't any issues like you describe.
Check the logs for any indication of errors, try to clear all custom settings, always use the default (not custom) presets until the issue is resolved.