ramapcsx2/gbs-control #72: Old version picture bleeding test is better.

tomudo, 2019-07-19:
I compare commit 523ca42a2084f25b09a514abf0017d63e54a1dd6 with latest. I use 240p test patterns in picodrive emulator in PS2. When I test by test patterns -> color bleed check, the old version color is sharp but latest version is blur.

ramapcsx2, 2019-07-19:
I'm often changing the various filters when I discover new things.
The current setup prioritizes jailbar and ghosting reduction over static test picture results.
This approach works better when playing games, but it isn't set in stone yet.

Some day I can imagine deciding on possible options for this, and then I'll make them available as user options.
It's a lot of parameters to consider, especially all the different hardware variations, so it's not simple.

ramapcsx2, 2019-07-19:
Small list of filters and other factors:
- IF LPF or Interpolator select: 1_02 1
- IF LPF/Interpolator bypass: 1_02 2
- IF unknown "TAP11" LPF bypass: 1_02 4
- IF U/V delay: 1_02 5/6
- IF data reduction (currently div 2)
- VDS U/V delay: 3_24
- VDS "Tap6" bypass: 3_24 3
- VDS clock interpolators 1 and 2: 3_40
- Enhancements: step response (3_2B), peaking (3_4E 0)
- sampling clock
- display clock

ramapcsx2, 2019-07-24:
@tomudo 
What output mode did you test this in?
I'm changing the mode for 480p once more. The delays were not correct.

tomudo, 2019-07-25:
It tested with NTSC mode.

I just tested 480p but picture of bleeding test look similar.

ramapcsx2, 2019-07-25:
Okay, that's odd then. NTSC (or PAL) modes either interlaced or not look correct on my PS2.
I even use Picodrive now :p 

One more question:
Are you use Component cables, or RGB?

tomudo, 2019-07-26:
I use RGBS with citrus3000 modded and Coponent cable, however, both have the same result.

I just swap to Aug 20,2018 firmware to test again with all available mode (960p, 720p,480p). The vertical lines are sharper and have all same width in bleeding test. Then I upload latest Jul 25,2019 firmware, the vertical lines are not sharp as old firmware and line width is not equally width. 

The blue lines have the most noticable problem, red and green have less noticable, white line has the least.

The pattern looks like these,
sharp and normal width- blur and fat - blur and fat - sharp and normal width ...... (repeated)

ramapcsx2, 2019-07-26:
Hm, could you take a picture of the effect? I'm not sure we're looking at the same test.

tomudo, 2019-07-26:
Picture from test of latest firmware: https://imgur.com/kumQ9Yj

Picture from test of old firmware: https://imgur.com/3jAgfEJ

ramapcsx2, 2019-07-26:
Okay, this is fine.
It is missing the 3_24 VDS Tap6 filter.
Enabling the filter will "fix" that issue, but comes with drawbacks such as pronounced jailbars and scrolling artefacts.

If you leave that test screen and run any regular game, the graphics will be good.
You can toggle the filter as well, if you want (t3t24t3), but keep the drawbacks in mind.

tomudo, 2019-07-26:
I tested filter and have result as you mention. Thank you for your help. 

Should I close this issue?

ramapcsx2, 2019-07-26:
Alright :)
The issue can stay open. It reminds me to add some form of user control for the filters.

ramapcsx2, 2019-07-26:
Update:
I checked just the effect of this filter on 3 boards.
Sometimes it tends to increase jailbars and shadows while scrolling, sometimes it doesn't.
It really depends a lot on how well the sampling clock (thus, the preset) fits a particular source.

This filter definitely will be a user option soon! :)

ramapcsx2, 2019-07-31:
Option is included now.
I've also defaulted the filter to enabled, as it was before.

closed 2019-08-19 by ramapcsx2