ramapcsx2/gbs-control #73: 1600x900 preset request

ramapcsx2, 2019-07-22:
tomrev over at shmups:
Hi, rama. May I ask for support 1600x900 monitor?

It's probably not going to be nice for NTSC (next even scale is x960), but maybe okay for PAL (x864).

tomudo, 2019-10-14:
NTSC PS2 games use 512×224 , 512×448 , 640x448. Vertical 2x and 4x scales are still less than 900p. Could it will be nice if have 1600x900 for PS2 system?

ramapcsx2, 2019-10-14:
So these are all variations of the common 224 line count retro resolutions, which indeed fit within a 900 line output.
The other common source line count is 240, which would not fit.  

Many games and systems don't make use of all 240 lines however, and are more closer to 224.  
Sometimes they call this avoiding overscan.  
For those situations, a preset with 224*4 = 896 would be ideal.  
Since that's just 4 lines short of 900, I think it'd make good sense to do a 900 line count preset :)  

As typical for gbscontrol, the actual horizontal resolution will be whatever fits the hardware best and looks sharpest.