ramapcsx2/gbs-control #74: Possible to connect Arcade/JAMMA PCB to GBS?  (re: RGB @ TTL voltage)

sofakng, 2019-07-29:
I've read the [Arcade Board](https://github.com/ramapcsx2/gbs-control/wiki/Arcade-Boards) wiki article and also the [related issue](https://github.com/ramapcsx2/gbs-control/issues/28) but I am still a little confused.

Is it safe to connect an arcade PCB (i.e. JAMMA) that outputs RGB @ TLL (as high as 5v p-p) to the GBS-8200?  If so, what is the output voltage with the potentiometer set to the far left (as described in the wiki article) ?

My goal is to reduce the TTL voltage to CGA/VGA level (0.7v p-p) and passthrough the RGB exactly as it is.  (i.e. no scaling, 15 kHz, etc)  I'm going to connect a VGA (RGBHV)  to Component video transcoder after the GBS-8200.

ramapcsx2, 2019-07-29:
I suppose it will work.
Start with the 3 potentiometers to the far right though, that's the largest attenuation.

You want the passthrough mode then, but keep in mind that it's still going to be sampled video.
The processing chain will be:
ADC > HD Bypass unit > DAC
The output will be RGBHV at TV levels.

The GBS has Component output capability, by the way.
The software isn't well suited for that yet though, as it's a niche use case so far.

sofakng, 2019-07-29:
Thanks so much for the information.  I was afraid it would be sampled video but it should be OK for now.

Can you explain a little bit more about what the potentiometers are doing and what it's attenuating?

If I understand correctly, VGA/TV voltage levels are 0v - 0.7v.  If the voltage is 0.7v then it's considered "full brightness".  (i.e. 0.7v RED would be fully red)

What are the potentiometers actually adjusting?  Does it reduce the input voltage before the processor does anything or does it adjust the output voltage?

ramapcsx2, 2019-07-29:
They form a voltage divider in series before the AC coupling capacitor into the ADC channel.
As they're turned right, the signal sees an increasing series resistance and at the same time a decreasing resistance to ground (a voltage divider).

0.7V is TV / regular VGA specs. Almost everything analog video uses these levels.

closed 2019-07-31 by ramapcsx2