ramapcsx2/gbs-control #75: Enable both Frame time lock + Sync watcher = interval blackout.

tomudo, 2019-07-31:
I want to enable frame time lock because I want to get rid of screen tearing. I must disable sync watcher to enable frame time lock that the screen won't show black screen but it does not convinent to disable sync watcher every time I restart system. Could this problem be fix?

ramapcsx2, 2019-07-31:
Constant black screening means that your display doesn't work with the FTL method.
Have you tried the other method yet? (There are 2 available.)

The problem shouldn't be with syncwatcher. It's odd that FTL works for you with it off.
Maybe what happens is that your display works fine with some timings, and doesn't work with the rest.
Syncwatcher triggers a routine that brings the output horizontal timings very close to the input's.
This could be the reason why it starts working when you disable syncwatcher.

tomudo, 2019-07-31:
I tried 2nd method but still have black screen problem.  :\

ramapcsx2, 2019-07-31:
Then your display doesn't have enough margin to use this method, at least not for this combination of source, video mode and output preset.

I'm interested in solutions for the problem, but so far no luck.
But I can say that many VGA to HDMI dongles work well with this.

ramapcsx2, 2020-02-07:
Hey, if you're still having this issue, there is now a hardware modification available that fully fixes it.   
See these 2 posts:   
https://shmups.system11.org/viewtopic.php?p=1396532#p1396532   
https://shmups.system11.org/viewtopic.php?p=1396777#p1396777

closed 2020-02-13 by ramapcsx2