ramapcsx2/gbs-control #77: Possibility of outputting interlaced modes?

pullasuti, 2019-08-08:
I am having a hard time getting my AVR to accept a signal from the GBS over component video. It seems to only support 480i or 576i. Is there any change of converting a 240p signal to 480i? 

But in any case: what a great project! :)

ramapcsx2, 2019-08-08:
Thanks! :)

The chip can do it but I've never attempted it.
Have you tried the passthrough option? If your source is 15kHz, it might work that way.

pullasuti, 2019-08-08:
Yup. 

Been trying passthrough with an Amiga and a Megadrive with no luck. The AVR is just being very picky (Denon X3500H, would really expect a little more from it but nope... even a cheap supermarket UHD seems like a better option for component video:).

ramapcsx2, 2019-08-09:
Just to make sure, have you tested your Component out on other sources / verified it works?
I can't vouch for the option much, since I never use it. Not enough time to cover all cases, unfortunately.

closed 2020-02-07 by ramapcsx2