ramapcsx2/gbs-control #78: The motion adaptive deinterlace is blur and dull color while scorlling

tomudo, 2019-08-14:
I notice in motion adaptive deinterlace while scrolling , the background is blur and darker, compare to bob deinterlace the background is not change in color or sharpness while scrolling. They're compared by playing dragon warrior 3 in NES emulator of PS2.

Is it normally effect of motion adaptive?

ramapcsx2, 2019-08-14:
Yeah, probably. I'm surprised you even noticed this.
There is a more elaborate deinterlacer method available that includes (more?) UV data.
This one requires another entire frame of history though, so it would increase lag a lot, so that'd be lots of work to get it running, for (probably) little benefit and additional lag.

tomudo, 2019-08-14:
I play several PS2 game with motion adaptive but could notice that effect. I think it may noticable when simple 8 bit era games like background pattern but complex background in most PS2 or modern games won't be noticed. 

I like almost lag free of gbs-control.

ramapcsx2, 2019-08-14:
Ah yeah, all deinterlacers have trouble working with sources that are 240p originally, but output in 480i.
It defeats some of the assumptions on pixel motion in-between fields.
I haven't seen your test, but in the 240p (p! :p) test suite, it can be seen in the red selection test when you move it.
The contours of the red in the active selection are less vivid than with 480i mode disabled.

Well, I want to balance the bad effects of deinterlacing with it's main benefit of image stabilization.
It's always going to show some problems, I think.

closed 2019-08-19 by ramapcsx2