ramapcsx2/gbs-control #79: Possibility for 1024x768 preset for old LCD/CRT?

Miggas, 2019-08-21:
Title says everything.
Also, will 480i have support for scanlines in the future?

ramapcsx2, 2019-08-25:
Another output resolution is always possible, but I have to limit this by how much I can support.
Each preset requires constant maintenance, so I limit them to the most useful ones.
1024x768 doesn't sound like it would scale particularly nice, and not a lot of displays require this / can't work with the other resolutions.

If anything, I would go for 1366x768, as that is a somewhat popular resolution.

Scanlines the way I do them can only work with 240p content. This is a limitation of the hack I use.

shift838, 2019-09-04:
There are quite a few old computers like the Geneve 9640 that could benefit from the 1024 x 768 option.

ramapcsx2, 2019-09-04:
@shift838 You probably mean support as an input? If so, that's already included.

ramapcsx2, 2019-10-13:
merged with the other preset request

closed 2019-10-13 by ramapcsx2

lovrenco, 2021-01-29:
@ramapcsx2 Hey, @Miggas had good question, I am testing device with various 15 inch CRT/LCD monitors, it has many issues. And, as you know, lower res monitors are way much better for retro gaming. 
Any possibility for support here?

gingerbeardman, 2021-01-29:
I use 640x480 LCD (now that's what I call low resolution) it works a treat. Maybe look to see if there are any of those for sale near you?