ramapcsx2/gbs-control #80: Compile errors

shift838, 2019-08-29:
trying to compile the code with all the required libraries and the correct selected board.  I get quite a few 'warning unused parameter' errors.  it does finishing compiling.  Will this be an issue or is this expected?

the below are just a few:

In file included from C:\Users\cvsch\Documents\Arduino\libraries\ESPAsyncWebServer\src\ESPAsyncWebServer.h:457:0,

                 from C:\Users\cvsch\Documents\Arduino\libraries\ESPAsyncWebServer\src\WebResponses.cpp:21:

C:\Users\cvsch\Documents\Arduino\libraries\ESPAsyncWebServer\src\AsyncWebSocket.h:86:10: warning: unused parameter 'i' [-Wunused-parameter]

     void operator ++(int i) { _count++; }

          ^

C:\Users\cvsch\Documents\Arduino\libraries\ESPAsyncWebServer\src\AsyncWebSocket.h:87:10: warning: unused parameter 'i' [-Wunused-parameter]

     void operator --(int i) {  if (_count > 0) { _count--; } ;  }

          ^

Nuck-TH, 2019-08-29:
You should ask ESPAsyncWebServer developers about this. Path pretty clearly shows that warnings occur not in gbs-control code.

ramapcsx2, 2019-08-29:
Yep, it'll be fine.  
The entire Arduino / ESP8266 environment is in constant development.  
Sometimes developers leave things unfinished, which can lead to warnings like these.

closed 2019-08-29 by ramapcsx2