ramapcsx2/gbs-control #81: no screen output

shift838, 2019-09-04:
i found this project last week and it interested me to be able to try and clean my video up for my Geneve 9640 (outputs 15kHz)that is connected via RGB to the GBS8200 then out to VGA.

I have built the board based with a NodeMcu ESP8266 D1 Mini and followed the Wiki for hardware setup and uploading of the code.

I am using the standard Sync Stripper circuit with the LM1881

[i found this project last week and it interested me to be able to try and clean my video up for my Geneve 9640 (outputs 15kHz)that is connected via RGB to the GBS8200 then out to VGA.

I have built the board based with a NodeMcu ESP8266 D1 Mini and followed the Wiki for hardware setup and uploading of the code.

I am using the standard Sync Stripper circuit with the LM1881

[i found this project last week and it interested me to be able to try and clean my video up for my Geneve 9640 (outputs 15kHz)that is connected via RGB to the GBS8200 then out to VGA.

I have built the board based with a NodeMcu ESP8266 D1 Mini and followed the Wiki for hardware setup and uploading of the code.

I am using the standard Sync Stripper circuit with the LM1881 as followed below:

http://etim.net.au/temp/forum/scart_cables/lm1881_datasheet_circuit.png

I am powering it with a USB cable for testing purposes.

However I get a complete blank screen.  I switched the SDA and SCL lines just in case but no difference.  Disable the GBSControl and it all works when it uses the internal GBS8200 firmware.

My serial monitor shows:

starting
(WiFi): still connecting..
<reset>
Chip ID: 1A 1
 G: 3B
 R: 3B R: 3C R: 3D R: 3E R: 3F
 B: 3B B: 3C B: 3D B: 3E B: 3F B: 40 B: 41
<reset>
<reset>
Scanning inputs for sources ...
(WiFi): STA mode connected; IP: 10.0.0.54
(WiFi): Access 'http://gbscontrol:80' or 'http://gbscontrol.local' (or device IP) in your browser
Activity detected, input: RGB
VSync: present
HSync: present
RGB/HV bypass 
ADC offset: R:3F G:3B B:41
RGB/HV bypass on
```````````````<reset>

Any clues?

shift838, 2019-09-04:
I launched the web ui and tried to toggle the RGBHV / component input.  I will get an Output Format: component then toggling back for RGBHV I receive an error of 'Unknown Timing!'

ramapcsx2, 2019-09-04:
The toggle is for output.

Well, I don't see a problem in the logs:
The device is getting detected as RGBHV and it starts in bypass mode.
If that's all the logs say, then the problem is with the receiving display.

In the sketch, around line 4849, set "preferScalingRgbhv" to 1.
uopt->preferScalingRgbhv = 0; // test: set to 1

shift838, 2019-09-04:
changed and recompiled/upload.

<reset>
ws16 connected
VSync: present
HSync: present
RGB/HV bypass 
ADC offset: R:3F G:3B B:41
RGB/HV bypass on

let me try a different monitor

shift838, 2019-09-04:
same issue with different monitor.  do I need the 100ohm on the sync in?

shift838, 2019-09-05:
since this is an old computer (80's) that uses RGB i don't think i need the 100ohm or 75 ohm within the circuit.  it works without the GBSControl firmware as I have it wired, but with the firmware no picture.

my standard GBS wiring working without GBSControl is:

RED > Red
GREEN > GREEN
BLUE > BLUE
GROUND > GROUND (all tied together)
GBS 5v power to LM1881 pin 8 VCC
LM1881 VSync (pin 3) > Vsync
LM1881 composit Sync Output (pin 1) > 470ohm > GBS8200 HSync 
computer CSYNC input > .1uf > LM1881 Composite Input (pin 2)

i'm thinking maybe wiring is off using the Mini D1  or maybe it's not compatible to use with my old Myarc Geneve 9640 computer although it is a 15kHz system.  We can use commodore amiga monitors on it.

The ESP8266 board I have is:

https://www.ebay.com/itm/D1-Mini-NodeMcu-4M-bytes-Lua-WIFI-Development-Board-ESP8266-by-WeMos/382553549033?ssPageName=STRK%3AMEBIDX%3AIT&_trksid=p2057872.m2749.l2649

shift838, 2019-09-05:
i finally got an output.  it seems the video signal is going in and out.  also look at the bottom of the screen of the video garbage.  it moves down the screen constantly.  it will go up to the top of the screen.  The resetting of the pic is about every 10 seconds.  the more it resets the more the color goes purple.

[image]

ramapcsx2, 2019-09-05:
It's still running RGB bypass mode. This is a problem on my end.
It's difficult to use separate (H + V) sync with the regular CSync code path.

If you have a CSync signal available, that will be easier to use.

shift838, 2019-09-05:
the original c-sync signal does come from my Geneve computer.  

how would I need to wire it up with the Csync then?

shift838, 2019-09-05:
[image]

ramapcsx2, 2019-09-05:
Ah, great.
Just go from your CSync pin to a 470 Ohm series resistor to the "S" pin on the GBS.
The resistor attenuates the CSync signal, the GBS does the sync extraction.
You can remove the LM1881 / disconnect it.

shift838, 2019-09-05:
that seemed to have done the trick!

the monitor I have hooked it up requires 1024x768 @ 60hz, so my monitor is reporting not optimum mode since it's set to something to a higher resolution.  I had to set it to the 640x480..

shift838, 2019-09-05:
is there a way to get the unit to power up at 640x480 or add a 1024x768?

ramapcsx2, 2019-09-06:
Lol, where do all these 1024x768 max monitors come from all of a sudden :D 

Once you selected an output preset on the web ui, it becomes the startup default.
So all you need to do is select 640x480 and restart.

ramapcsx2, 2019-10-11:
Initial issue solved

closed 2019-10-11 by ramapcsx2