ramapcsx2/gbs-control #83: translate to Portuguese

ramirosoad, 2019-10-09:
Hello, I did the translation of the HTML file to Portuguese, but when I put in ESP it still in English, could you help me?
Congratulations on the project, it is getting better.

ramapcsx2, 2019-10-09:
Hi,
the .html file needs to be converted to a header (.h) file, so it can be embedded with the rest of the program.  
I use Linux tools to do this command chain in a terminal:
`gzip -c9 webui.html > webui_html && xxd -i webui_html > webui_html.h && rm webui_html && sed -i -e 's/unsigned char webui_html\[]/const uint8_t webui_html[] PROGMEM/' webui_html.h && sed -i -e 's/unsigned int webui_html_len/const unsigned int webui_html_len/' webui_html.h`

You need a Linux environment (virtual machine or Cygwin) with these tools installed:  
gzip, xxd, sed

ramirosoad, 2019-10-11:
It worked here, thank you very much

ramapcsx2, 2019-10-11:
Good to hear :) 
I don't plan on maintaining translations by the way. I hope you understand.

closed 2019-10-11 by ramapcsx2