ramapcsx2/gbs-control #86: composite/s-video support through component video

sethenstien, 2019-10-24:
is there anyway to possibly input composite or s video though the component inputs, that isn't monochromatic? 

also has anyone tried one of the $15ish dollar s-video/composite to vga with these. I've heard those have issues putting out older signals in mono. i doubt it can currently get composite or s-video any other way but if there is tell me.

ramapcsx2, 2019-10-24:
To restore / decode color out of these signals, a decoder is required. The scaler chip doesn't have a decoder, so it can't be done that way with existing hardware.

External "to VGA" devices have mini upscalers in them that destroy the quality before it gets to gbscontrol.
The only thing that would work is a gaming focused external device, such as the Retrotink 2x.

eadmaster, 2021-03-09:
What about decoding directly on the ESP8266 via an analog pin, or using a [separate digital video decoder](https://github.com/wooky/esp32-analog-video-over-wifi)?

I think this would make a nice addition to the project.

ramapcsx2, 2021-03-09:
If I read this right, they're suggesting a digital Composite decoder, outputting to an ESP32 via I2S.
From there on they manipulate the now digital data to send it over WiFi.
The step to get the digital line data into the GBS scaler IC is missing.

adwil13, 2022-05-27:
Any ideas if it's even possible? I just put together my GBS-C and trying to connect my NES to it, but due to chip shortages any sensible option to transcode composite to component (Koryuu Transcoder) or to mod my NES (NESRGB) are constantly sold out everywhere.
RetroTINK 2X is not really a solution as it outputs digital signal over hdmi so it skips GBS-C altoghether and their deinterlacing algorithm is just inferior to GBS-C motion adaptive one.