ramapcsx2/gbs-control #87: Board Adapter not working with internal VCC

leonpierre, 2019-11-04:
I build the adapter for the ESP8266:

[image]

and nothing happens (no WIFI ...) but if I connect through Micro-USB I can connect and
also see this output from Serial Monitor:

```
startup
SPIFFS mount failed! ((1M SPIFFS) selected?)
(WiFi): AP mode (SSID: gbscontrol, pass 'qqqqqqqq'): Access 'gbscontrol.local' in your browser
<reset>
Chip ID: 1A 1
 G: 3E G: 3F G: 40 G: 41
 R: 3E R: 3F R: 40 R: 41 R: 42 R: 43 R: 44 R: 45 R: 46 R: 47 R: 48 R: 49
 B: 3E B: 3F B: 40 B: 41 B: 42 B: 43 B: 44 B: 45
<reset>
<reset>
Scanning inputs for sources ...
```

Checks:

- 3.3V measured on ESP8266
- Jumper set
- Reset Button only works if connected over USB

Any hint what could be wrong?

Thx

PS: C11 currently not removed.

0xfee15bad, 2019-11-04:
The VCC of the GBS board is indeed at 3.3V, and you should feed that to one of the 3V pin of the ESP8266 board.
Since the VIN pin on the ESP8266 is regulated I'm not sure if it can take 3.3V, to use that pin it's better to have it connected directly to the main power supply of the GBS board by using the two pins (P9 on the PCB) next to your power connector, that's what most of us do and it is the recommended method given in the wiki.

tomudo, 2019-11-05:
Could you measure voltage with voltmeter between VIN and G on ESP8266 while GBS powered?

tomudo, 2019-11-05:
-

tomudo, 2019-11-05:
https://github.com/nodemcu/nodemcu-devkit-v1.0/issues/15

Vin should be between 4.75 - 10V

ramapcsx2, 2019-11-05:
Right, the main issue is that the ESP8266 board has no power.  
Please disconnect all 3.3V lines from the ESP8266 and connect ground and Vin to the GBS power input.  
The 2 pin cable that you got with your GBS is ideal for that.  

Next, please check your Arduino IDE setting and fix the SPIFFS partition. See here:  
https://github.com/ramapcsx2/gbs-control/wiki/Software-Setup#esp8266-ide-settings  

Next, please adjust the 3 RGB potentiometers all the way to the left.  

Good luck!

leonpierre, 2019-11-09:
Thx! It works now like a charm! ;-)

ramapcsx2, 2019-11-09:
Great :)

closed 2019-11-09 by ramapcsx2