ramapcsx2/gbs-control #88: RGB input

tomudo, 2019-11-09:
I use RGBs input via VGA port. When save preset it show message "open save file failed!"

In function savePresetToSPIFFS when rto->videoStandardInput value is 14, it not save preset.

ramapcsx2, 2019-11-09:
Yeah, modes 14 and 15 are for VGA / PC sources.
Mode 14 used to be straight RGB bypass, but now with the full processing path, it doesn't have all the required logic to save presets.

I need to think about this.
The problem is that your PC source could present all kinds of formats and resolutions, all the custom preset code is geared towards single / fixed formats though.

tomudo, 2019-11-09:
My source input is from modded PS2 that output RGB composite sync. It does not really VGA which is RGB HV.

ramapcsx2, 2019-11-09:
Okay, that's "regular" RGB + CSync then and it should never go into mode 14.
Mode 14 / 15 should only activate if there is an active VSync signal on the VS pin.
This is only possible with a source that has HSync and VSync separately.

tomudo, 2019-11-09:
I checked the cable and quite sure that nothing wrong with wiring. Only 1 csync signal pin connect to H sync pin (middle of lower pin). V sync pin of VGA has nothing connect. 

Old 18/08/19 commit has no saving problem.

ramapcsx2, 2019-11-25:
Is it still doing this with the latest version?

tomudo, 2019-11-26:
Commits on Nov 24, 2019 works. Thank you.

closed 2019-11-26 by tomudo

ramapcsx2, 2019-11-26:
Okay, good to know, but it's still weird that it happened.
I think there might be an edge case that I never run into when testing..

If the bug returns, please tell me :)