ramapcsx2/gbs-control #89: "Build the Hardware" has confusing instructions

Cybolic, 2019-11-19:
The instructions has the following for Wemos D1 boards:

> Wemos D1: SDA to D14, SCL to D15

From what I could tell, none of the instructions/manuals for the Wemos refer to the pins as D14/D15 though it is printed on one version.
Shouldn't the instructions read the following?

> Wemos D1 R2 (blue): SDA to D4/D14 (GPIO4), SCL to D3/D15 (GPIO5)
> Wemos D1 R2 V2.1.0 (black) or Wemos D1 Mini: SDA to D2 (GPIO4), SCL to D1 (GPIO5)

Sorry if this is obvious, but it puzzled me quite a bit not finding any pins higher than D8 on the Wemos D1 boards I could find.

ramapcsx2, 2019-11-19:
Oh, it gets worse with each new variation of ESP8266 dev boards.   
Pin naming is definitely a weak point of the ESP family.   
https://872c4715dbe9f10b83d1-0b39dab3ba460c18aad59cf32aacf5c8.ssl.cf5.rackcdn.com/TE482-E-10-main3.jpg   

What seems to matter most is the function name (SCL, SDA) that's printed on the boards.   
I don't know how to best convey all that information, while *not* further confusing people.

ramapcsx2, 2019-11-19:
If your information on the "D1 R2 V2.1.0 (black)" is correct, I can certainly put this into the documentation, if you think it helps.

Shoometsu, 2019-11-19:
I've build a prototype PCB to connect both Wemos D1 R1/R2 and had to pay
attention between the pin difference in order to make it work properly too.

Cybolic, 2019-11-19:
The info on "D1 R2 V2.1.0 (black)" was just from looking at shop listings, but it seems to match what I can find: https://wiki.wemos.cc/products:d1:d1

How about the following instead?

> Connect the two I2C bus wires SDA (GPIO4) and SCL (GPIO5).
> Most boards seem to use D2 for SDA and D1 for SCL, but check your specific board before wiring.

You could also do a table, but then that would need to kept up to date as well.

ramapcsx2, 2019-11-22:
Better now? :)

Cybolic, 2019-11-22:
Excellent! A notice about the debug pin would probably be good as well - does it go to GPIO12 / MISO?

ramapcsx2, 2019-11-22:
I .. think it does :p 
It's called "D12/MISO/D6" on my board.
It's a big mess, honestly.

Cybolic, 2019-11-23:
Sounds like labelling it "MISO" in the wiki would be the way to go then. At least that's everything accounted for :)

ramapcsx2, 2019-11-23:
Updated.

Cybolic, 2019-11-23:
Perfect.

ramapcsx2, 2019-11-23:
Okay, hope it's easier for new people now :)

closed 2019-11-23 by ramapcsx2