ramapcsx2/gbs-control #91: Custom Resolution for Sega Model2 Arcade Boards

cglmrfreeman, 2019-11-20:
The Sega Model2 has an interesting display resolution as cited by [SegaRetro](https://segaretro.org/Sega_Model_2#Graphics):
> Display resolution: 496×384 pixels, 24 Hz HSync, progressive scan (non-interlaced), double-buffering
>Overscan resolution: 656×496

Attempting to adjust these on the native GBS-8220 (The model I own) is tedious and results in a poor picture each time the unit is powered on. I was wondering if this could be remedied by scaling to the correct resolutions natively using this CFW. The 680x480 option, frankly, it sucks. Much fine-tuning is needed with each boot and it usually ends up whitewashing out anyway.

I am hoping with adjustments (or special commands ala [TomDalby](https://tomdalby.com/other/gbs8200.html)`s5s12s7fjn zzzzz`) the picture could be radically improved by default and ready for capture. Also, I do not know if it's possible but the option of both a display and overscan resolution would really be nice.

Note: I will be using the [WeMos D1 Mini to GBS 8200 via PLCC socket rev2](https://oshpark.com/shared_projects/lrvd4065) by [jmuzi](https://shmups.system11.org/viewtopic.php?p=1341354#p1341354) and not the NodeMCU as used by TomDalby. This is a personal preference due to the footprint it leaves and the intriguing nature of how to attach the CFW.

ramapcsx2, 2019-11-21:
The cited values have some issues, but this is clearly one of the "medium res" / 24kHz modes.
I recently started supporting them.
Don't expect a perfect experience out of the box, but you should get good results with some tweaking.
Results can then be saved as a custom preset.

cglmrfreeman, 2019-11-21:
Great, once my parts arrive and everything is working I will update this issue with my results and let you know how it all works. If nothing is wrong, I'll close the issue!

cglmrfreeman, 2019-11-24:
Well, my PLCC IC sockets arrived and I ordered the wrong ones. This project I started is getting a little too expensive to continue. However, I did receive a 44-pin PLCC test clip for another project which fits on the MTV230M perfectly. Initial testing revealed heavy alteration of the picture was needed in order for a proper picture, but the picture does indeed look good. Scanlines, Picture Right, Border and Masking (all) don't have any affect on the picture. Picture Left seems to increase the left border into the picture rather than moving the picture to the left.

Another (minor) issue I encountered was after connecting the WeMos to my WiFi, it could not be accessed over Wifi. My router denied it WLAN access and it could only be accessed through wire connected devices. But that's most likely something on my end but the hours of troubleshooting couldn't solve it.

I also removed C11 as reccomended, but did not notice much noise removal. Unfortuinately, I burned off one of the pads removing the cap, so I can't put on the electrolytic as suggested as an alternative.

Overall, however, I am extremely pleased with the picture this produces. I kept looking at OSSC units, but they did not seem to have the level of customization this has for my specific circumstances surrounding the M2B's resolution. Closing this (non)issue now. Thank you for your assistance.

closed 2019-11-24 by cglmrfreeman

ramapcsx2, 2019-11-24:
Sure, keep watching the project for updates regarding RGBHV or HD bypass. These are for the special modes, such as medium res.

Eventually, medium res will be able to use the default upscale processing path.
It's similar enough to regular EDTV 480p.

The WiFi issue sounds like it's fundamentally not working. Wrong key perhaps.