ramapcsx2/gbs-control #93: OTA Updates

TioChico27, 2019-11-27:
Hello!

Are OTA updates already working? Because i can't make it work.

Amazing project by the way.

Regards.

Nuck-TH, 2019-11-27:
Press enable ota in web interface and choose gbs adress in Tools - Board in Arduino. Then you can upload the sketch. However, even while there is Restart option in web interface, i recommend to power cycle the board.
That's it.

TioChico27, 2019-11-27:
> Press enable ota in web interface and choose gbs adress in Tools - Board in Arduino. Then you can upload the sketch. However, even while there is Restart option in web interface, i recommend to power cycle the board.
> That's it.

Thanks for reply.

It's working now!

Regards.

closed 2019-11-27 by TioChico27