ramapcsx2/gbs-control #94: Output is always passed through, never scaled

stephenrkell, 2019-12-12:
Thanks for your work on this great project. I recently built the hardware and installed the software with apparently no problems, and got everything wired up to my GBS 8220 v3. Unfortunately I can't seem to get it to do any scaling. I suspect it might be my debug pin not being properly soldered... but please see if any other explanation makes sense.

When I power or restart on the device, I get an output VGA signal that is just a passed-through input signal, i.e. the same resolution (720x400 VGA text).
```
Restart
(WiFi): IP: 192.168.128.136 Hostname: gbscontrol.local
preset preference = 0
component output = 0
pal force 60 = 0
matched = 0
<reset>
Chip ID: 1A 1
<reset>
<reset>
Scanning inputs for sources ...
```

Then if I turn on my VGA input, 720x400 70Hz (it's a PC booting, so includes some extraneous resets):

```
Activity detected, input: RGB
VSync: present
HSync: present
RGB/HV bypass 
ADC gain: reset
ADC offset: R:41 G:41 B:40
h: 511 v:1540 PLL:00 A:484848 S:02.08.00.00 I:83 D:00a7 m:0 ht:1726 vt:  0 hpw:   1 s: 1 u: 0 s: 2 W:-45
h: 511 v:1540 PLL:00 A:484848 S:02.08.00.00 I:83 D:00a7 m:0 ht:1726 vt:  0 hpw:   1 s: 1 u: 0 s: 2 W:-45
``(H-PLL) state: 1
h: 511 v:1540 PLL:00 A:484848 S:02.08.00.00 I:00 D:00a7 m:0 ht: 937 vt: 58 hpw:   1 s: 0 u: 0 s: 2 W:-44
h: 511 v:1540 PLL:00 A:484848 S:02.08.00.00 I:00 D:00a7 m:0 ht: 983 vt: 58 hpw:   2 s: 0 u: 0 s: 2 W:-45
`(H-PLL) state: 2
h: 511 v:1540 PLL:00 A:484848 S:02.08.00.00 I:00 D:00a7 m:0 ht:1432 vt:338 hpw:   2 s: 0 u: 0 s: 2 W:-42
h: 511 v:1540 PLL:00 A:484848 S:02.08.00.00 I:00 D:00a7 m:0 ht:2905 vt:338 hpw:   1 s: 0 u: 0 s: 2 W:-42
`(H-PLL) state: 1
h: 511 v:1540 PLL:00 A:484848 S:02.08.00.00 I:00 D:00a7 m:0 ht:1024 vt:432 hpw:   2 s: 0 u: 0 s: 2 W:-41
h: 511 v:1540 PLL:00 A:484848 S:02.08.00.00 I:00 D:00a7 m:0 ht:1024 vt:438 hpw:   2 s: 0 u: 0 s: 2 W:-45
`(H-PLL) state: 3
h: 511 v:1540 PLL:00 A:484848 S:02.08.08.00 I:00 D:02a7 m:15 ht:  79 vt:521 hpw:   1 s: 0 u: 0 s: 2 W:-41
h: 511 v:1540 PLL:00 A:484848 S:02.08.08.00 I:00 D:02a7 m:15 ht:1065 vt:521 hpw:   3 s: 0 u: 0 s: 2 W:-41
`(H-PLL) state: 2
h: 511 v:1540 PLL:00 A:484848 S:02.08.08.00 I:00 D:02a7 m:15 ht:1939 vt:256 hpw:   2 s: 0 u: 0 s: 2 W:-44
h: 511 v:1540 PLL:00 A:484848 S:02.08.08.00 I:00 D:02a7 m:15 ht: 992 vt:256 hpw:   3 s: 0 u: 0 s: 2 W:-45
```(H-PLL) state: 3
h: 511 v:1540 PLL:00 A:484848 S:02.08.08.00 I:00 D:02a7 m:15 ht:3355 vt:244 hpw:   3 s: 0 u: 0 s: 2 W:-42
h: 511 v:1540 PLL:00 A:484848 S:02.08.08.00 I:00 D:02a7 m:15 ht:2846 vt:277 hpw:   2 s: 0 u: 0 s: 2 W:-44
``(H-PLL) state: 2
h: 511 v:1540 PLL:01 A:484848 S:02.08.08.00 I:00 D:02a7 m:15 ht:1856 vt:509 hpw:   3 s: 0 u: 0 s: 2 W:-44
h: 511 v:1540 PLL:01 A:484848 S:02.08.08.00 I:00 D:02a7 m:15 ht:1856 vt:503 hpw:   3 s: 0 u: 0 s: 2 W:-44
`(H-PLL) state: 3
h: 511 v:1540 PLL:01 A:484848 S:02.08.08.00 I:00 D:02a7 m:15 ht:1861 vt:452 hpw:   3 s: 0 u: 0 s: 2 W:-41
h: 511 v:1540 PLL:01 A:484848 S:02.08.08.00 I:00 D:02a7 m:15 ht:1856 vt:453 hpw:   1 s: 0 u: 0 s: 2 W:-45
`(H-PLL) state: 2
h: 511 v:1540 PLL:03 A:484848 S:02.08.08.00 I:00 D:02a7 m:15 ht:1856 vt:517 hpw:   4 s: 0 u: 0 s: 2 W:-42
h: 511 v:1540 PLL:04 A:484848 S:02.08.08.00 I:00 D:02a7 m:15 ht:1856 vt:519 hpw:   4 s: 0 u: 0 s: 2 W:-42
`(H-PLL) state: 3
h: 511 v:1540 PLL:04 A:484848 S:02.08.08.00 I:00 D:02a7 m:15 ht:1856 vt:519 hpw:   3 s: 0 u: 0 s: 2 W:-43
h: 511 v:1540 PLL:04 A:484848 S:02.08.08.00 I:00 D:02a7 m:15 ht:1856 vt:513 hpw:   3 s: 0 u: 0 s: 2 W:-45
`<reset>
Activity detected, input: RGB
VSync: present
HSync: present
RGB/HV bypass 
ADC gain: reset
ADC offset: R:41 G:41 B:40
h: 511 v:1003 PLL:04 A:484848 S:02.08.00.00 I:c3 D:00a7 m:0 ht:1114 vt:516 hpw:   3 s: 1 u: 0 s: 2 W:-44
h: 511 v:1003 PLL:06 A:484848 S:02.08.00.00 I:c3 D:00a7 m:0 ht:1840 vt:515 hpw:   4 s: 1 u: 0 s: 2 W:-44
`````````````(SP) HS active low
(SP) VS active low
(H-PLL) state: 2
h: 511 v:1003 PLL:06 A:484848 S:02.08.0a.00 I:00 D:0625 m:15 ht:1856 vt:448 hpw: 222 s: 5 u: 0 s: 2 W:-44
h: 511 v:1003 PLL:07 A:484848 S:02.08.0a.00 I:00 D:0625 m:15 ht:1856 vt:448 hpw: 222 s: 5 u: 0 s: 2 W:-43
(SP) VS active high
```

but I'm still seeing the passed-through input only, no scaling. If I click on the 1280x960 preset, I get the following output. 
```
RGB/HV bypass 
ADC gain: reset
ADC offset: R:41 G:41 B:40
h: 511 v:1003 PLL:08 A:484848 S:02.08.06.00 I:03 D:04a4 m:15 ht:1856 vt:448 hpw: 222 s: 1 u: 0 s: 2 W:-48
h: 511 v:1003 PLL:10 A:484848 S:02.08.06.00 I:03 D:0425 m:15 ht:1855 vt:448 hpw: 222 s: 1 u: 0 s: 2 W:-48

```

... but no change in output resolution. It seems to be the same no matter which preset I choose. (If I change the input resolution, the output resolution changes to match it.)

Some dumps: System -> List Options says

```
/preset_unknown.2 1732
/preferencesv1.txt 14
preset preference = 0
frame time lock = 0
preset slot = 1
frame lock method = 0
auto gain = 1
scanlines = 0
component output = 0
deinterlacer mode = 0
line filter = 1
peaking = 1
pal force60 = 0
matched = 1
6-tap = 0
```

and Development -> Get Video Timings says

```
HT / scale   : 0 0
HS ST/SP     : 0 0
HB ST/SP(d)  : 0 0
HB ST/SP     : 0 0
------
VT / scale   : 0 0
VS ST/SP     : 0 0
VB ST/SP(d)  : 0 0
VB ST/SP     : 0 0
IF VB ST/SP  : 0 0
```
... which seems wrong. Is this a sign that my soldered wire to the debug pin is not correctly soldered, perhaps?

ramapcsx2, 2019-12-13:
Hey,
first of all, the software seems very old.
Please install the latest master branch for much improved scaling RGBHV support.
https://github.com/ramapcsx2/gbs-control/archive/master.zip

Shoometsu, 2019-12-13:
Hi! I use GBS-control with an x68000, and it shows the same "RGB/HV bypass" message for 640p / 31Khz signals and my Screen says it's a constant 640p signal being receivd, but works with 240p / 15Khz -> 640p  without problems, it even enable scanlines.  Was it intended to work this way?

stephenrkell, 2019-12-13:
I can confirm that with the latest software, I do get scaling. Hooray! Thanks for that. It'd been longer than I realised.

(I have a minor issue with the left-hand few pixel columns being chopped off the output, in lower-res VGA modes, but glad to know you're actively working on the RGBHV support... I'll keep an eye on updates.)

ramapcsx2, 2019-12-14:
@stephenrkell 
Yeah, VGA modes are more difficult to work with, mainly due to their variable refresh rates.
I'll try to get the common DOS modes right at least though.

@Shoometsu 
Is 640p correct? It sounds like 800x600 or higher.
Modes above 640x480 are currently excluded from scaling, as supporting them adds complexity and they usually don't benefit from additional scaling.

closed 2020-01-02 by ramapcsx2