ramapcsx2/gbs-control #96: Adjust Htotal for active FTL?

goutsune, 2020-01-02:
Hello there, I'm actually using CRT monitors with GBS hardware and while testing FTL I noticed that I always get either jumping image or squishing image by about 1 line, depending on CRT monitor I use. However I also noticed that adjusting htotal by hand does not seems to noticeably affecy the picture, so I was wondering maybe it's a good idea to use it instead of vtotal?

goutsune, 2020-01-02:
I just quickly hacked together an option to adjust htotal, never-mind, it is still noticeable with 2 point adjustment. Perhaps more juggling with sync pulse and or front/back porches might alleviate this.

ramapcsx2, 2020-01-02:
I'll encourage you to experiment :)

ramapcsx2, 2020-01-12:
Just a heads up:  
The software targets an htotal value that's closer to the source now.  
I've disabled a round down step since it's no longer needed.  
As a side effect, the tear line shows up less often. When it does, it's also less noticeable and it even tends to disappear even before rolling off screen.

ramapcsx2, 2020-02-13:
No progress here, and the external clock generator option exists now.

closed 2020-02-13 by ramapcsx2