ramapcsx2/gbs-control #97: Enhancement Request: Sliders/Input Boxes for +/- values?

cglmrfreeman, 2020-01-05:
It seems I am adjusting my gbs-control (very) frequently to work with my Sega Model2 board since I began experimenting in Issue #91 

My problem is erratic and frequent disconnects when trying to adjust values over Wifi. Connecting the WeMos D1 to my network, as previously stated, blocks any wifi connected device from editing its settings. Not very helpful when the only wired computer in the house is far away from the desired gbs-control used location. My iPhone has never been able to connect and controls are difficult and often unresponsive on an Android phone.

I've managed to solve the issue of noise and other strange issues by bypassing my supergun's hardware and connecting the GBS directly to my arcade board. I'm close to perfect, but adjusting these settings has proven painful and the only after an "Erase Flash: All Flash Contents" upload of the software to the WeMos D1 allows a connection to its network. After approximately 100 modifications/clicks (pressing adjusting buttons takes a lot of work) the network begins disconnecting, refusing to connect to "gbscontrol.local" yet remaining connected to the wifi "network".

I'm at my wits end having to flash this over and over again to adjust settings and was wondering if perhaps I could input values instead of constantly editing them. Perhaps a tutorial on making my own preset? An option to upload "preferencesv2.txt" over the network or serial? I wouldn't even mind uncommenting code to define my own custom "uopt" struct as the default after flashing. I think sliders would be useful but unsure if the custom WebGUI can handle that, asyncing with the debug output... hmm.... 

Please let me know what you think to help assist in this situation, but it's by no means a priority.

Nuck-TH, 2020-01-05:
Try switching wifi channel in your access point. In some conditions scaler chip creates strong interference that clashes with wifi signal - so changing channel moves wifi frequency away from noise.

ramapcsx2, 2020-01-05:
Over time I've found many more references to the ESP8266 not tolerating electromagnetic field noise at all.  
I can only recommend some improvements:  
- heatsink replacement with a ceramic one  
- placing the ESP8266 below the GBS board (z-axis, so the GBS board needs to be elevated)
- using an ESP8266 with external antenna  

These all help with the problem and at some point, they'll be completely gone.

With regards to sliders in the web ui, those would be great! (That's a hint to you ;p)

cglmrfreeman, 2020-01-06:
What I ended up doing was purchasing a LOLIN D1 Mini Pro V2 and used an external antenna for now. Worked a treat! As for what you have in store, it'll come regardless if this issue is open or closed so, for now, I'm going to go ahead and close it as my primary issue has been solved!

closed 2020-01-06 by cglmrfreeman