ramapcsx2/gbs-control #98: GBS Control Not being picked up by phone WIFI

rabidgerry, 2020-01-28:
Ok so everything was fine regarding my phone detecting my gbs control, I was able to control from the url control panel.  However I found in order to load the page I had to turn my phones data.

Anyways in the control panel I went into the section that says system and went into the wifi settings.  Here I attempted to enter in my home wifi details.  After doing so suddenly gbs control is no longer detectable by my phone or anyone else's phone in my house.

How can I remove these details again so gbs control will be recognised by phone or pc (I would prefer not to have to restore to defaults if possible)?

Also why does this happen?  I thought the whole idea was that gbs control could use home wifi.

Thanks in advance.

ramapcsx2, 2020-01-29:
Sounds like the GBS is properly connected to your home network, and you just can't access it from there.
If you can, check your router interface and see if you can find the device there.  
It'll show the IP address as well, which you can use to connect directly.  
Sometimes home routers prevent connected devices from communicating with each other. You want to disable that in that case.

rabidgerry, 2020-01-29:
Ok so I was reading how I can first of all see what devices are on my network, and since I know I added my home network onto the gbs control I figured I might see it once I logged into my router.  So when I did this I of course saw the device and also the IP address.  So first time round without trying I anything else I typed the IP address into my browser on my phone and the control interface appeared immediately and allowed me to make adjustments.  So I did not have to enable anything, it was just a case of me finding out the IP of the gbs control.  To be honest, as a novice, I am not sure I would ever have known where to find the IP address so that was a fluke.  But regardless I am up and running with the interface and connection seems good.

ramapcsx2, 2020-01-30:
Great!  
I know about these IP troubles. They're the same for all projects that need to connect a microcontroller to a network.  
I've added 2 mechanisms that may work. Use either http://gbscontrol/ or http://gbscontrol.local/ .  
But both mechanisms can fail. It depends on the router and even your device's operating system.  
If they don't work, only directly using the IP address is possible, unfortunately.

closed 2020-02-01 by ramapcsx2